Ultimaker/Cura #2139: Model is not placed in center position but it can fit to the build volume

UM2EXTENDED, 2017-07-20:
Cura 2.6.2 OS X

In Cura 2.4.0 everthing was OK, but since I have updated to 2.6.2 the same model is located outside the bed if I open the file.

You can right click and put to center position, everthing is OK and can be printed just fine but you have to do it by yourself insted of automatically done by Cura.

Ghostkeeper, 2017-07-25:
I'm unable to reproduce this.

It always centres exactly in the middle of the volume. Sometimes that's not exactly the middle of the printable area if there are disallowed areas (such as a single-extrusion print with a printer that has a nozzle offset). And of course a 3MF file has an offset in the file, which would be intended.

What file was this happening with? Are you allowed to share that file?

UM2EXTENDED, 2017-07-25:
Sorry as it is for busniess use (prototype printing) I can't share the files with you.

I have different .stl files with following measurements:
196 x 196 x 50mm --> Auto position is at x200 / y100 / z0 (circle shape)
189 x 189 x 18mm --> Auto position is at x200 / y100 / z0 (circle shape)

I just double click the .stl file and it will open Cura automatically, the position is always at x200 / y100 / z0 by default.

Ghostkeeper, 2017-08-07:
I created a cube of 196x196x50mm using OpenSCAD.

When I load it into a (Custom FDM) printer that has sufficient size (230mm, to account for brim and such), it seems to load it in the centre (`[0, 0, 0]`). When I load it in a printer that is much too small, it loads it in the centre as well but doesn't align it with the build plate (`[0, 0, -25]`). When I load it in the Ultimaker 3 however, I reproduce your problem where it loads it on the side at position `[200, 100, 0]` for some reason. When I then centre the model, it still doesn't fit because of the nozzle offset.

In the third test, it should've loaded at the centre (`[0, 0, -25]`) and then don't align with the build plate because it doesn't fit.

nallath, 2017-09-05:
It places the model on the side due to the auto-arrange. It tries to put the model in the center of the build plate, which fails (due to the center not being the center of the allowed area). If an object can not be placed, it is moved to the right of the build plate.

Ghostkeeper, 2017-12-04:
This issue got removed from our planning by our project manger, because it's older than 12 weeks.

ianpaschal, 2018-07-30:
This no longer reproduces in Cura 3.4.1.

closed 2023-06-24 by github-actions[bot]