Ultimaker/Cura #2140: Save to Removable Drive only works once

archibaldjr, 2017-07-20:
After saving a file to a removable drive, the function is disabled. I have to close and reopen Cura in order to save another file.  I first noticed this in 2.6.1, and the problem is still in 2.6.2

AlucardF0X, 2017-07-24:
Yeah I've been getting this too.

The buttons don't work, I mash them like crazy and they don't do anything. I'm forced to use 'save to folder' and then manually eject. It was charming when it worked... 

What's weird is it worked for a long time in 6.1 and 6.2 then one day it just... stopped? It might be a driver issue silently erroring out.

nallath, 2017-07-25:
2.6.2 is no different from 2.6.1. The only difference is that 2.6.2 fixed a build issue for Mac.

archibaldjr, 2017-07-25:
Thank you Jaime.

 

I installed 2.5.0 and the problem went away, things behave as they should.

 

In 2.6.x, after I save my file to a removable drive, and the  removable drive is ejected, it is not possible for me to reconnect the removable drive and save again.

 

Bill

microdesigns2000, 2017-07-26:
Me too.  The save to flash button was nice when it worked.

Ghostkeeper, 2017-08-07:
Our testers have trouble reproducing this. What operating system are you on? What type of removable drive are you using? Is it an SD card reader? Does it go via a USB hub or a USB port on your monitor?

microdesigns2000, 2017-08-07:
Hi,

 

Thanks for looking into this issue! Here is my info.

 

System

OS: Windows 10 Home

PC: Lenovo Yoga 900-13ISK2

SD card reader: Built-in SD Card port

USB hub: No USB devices are used related to the removable drive

 

SD Card

SD card: Kingston 8GB 31560-016-A00LF (micro-SD), inserted into a micro-SD adapter

Micro SD adapter: Qumo MicroSD Adapter, no part number. 

 

SD Card: Sandisk Ultra Plus 16GB 71870DRFV00S

Micro SD adapter: Sandisk microSD microSDHC microSDXC Adapter, no part number.

 

To reproduce this problem

Open Cura with no SD card inserted [Cura only has greyed-out “Save to File” button].

Open an .stl and slice [Cura now has activated “Save to File” button].

Insert the Micro SD adapter with micro SD card into the built-in SD card port of the computer [OS beeps, Cura now has activated “Save to Removable Drive” button].

Click the “Save to Removable Drive” button [Cura responds with ‘Saving to Removable Drive…’, then Cura responds with “Saved to ].

After the save is complete, press the “Eject” button [Cura responds with “Ejected Removable Drive” and OS responds with beep.  Note: Cura maintains the activated “Save to Removable Drive” even though no drive exists.  This appears to be where the logic is not working properly and I would troubleshoot from this point.  Cura properly ejects, but ‘forgets’ to reset the available states of this button and drop-down].

Remove the SD card adapter from the computer [No response].

Insert the SD card adapter into the computer [OS beeps].

Press the “Save to Removable Drive” button again [No response].

Any attempt to switch between “Save to File” and back to “Save to Removable Drive” fails.

“Save to File” works properly.

 

Thanks again

fieldOfView, 2017-08-07:
What driveletter is your cardreader on?

archibaldjr, 2017-08-07:
Hi,

 

The issue seems to be solved in Cura 2.7 beta, FYI.

microdesigns2000, 2017-08-08:
It appears as drive E:\

 

Also, I downloaded 2.7.0 beta, and the problem does not occur.

 

Thanks!

microdesigns2000, 2017-08-08:
Yes, I installed 2.7.0 beta, and the problem is gone.  Thanks!

Ghostkeeper, 2017-08-23:
Fixed in #1022.

closed 2017-08-23 by Ghostkeeper