Ultimaker/Cura #2144: 2.6.2 Layers view options

PJParish, 2017-07-22:
Under preferences in 2.3.1 there is an option to show the top layer only.  While this seems a strange place to put that feature it is really useful and seems to be missing in 2.6.2.  I'm not finding that view mode anywhere.

fieldOfView, 2017-07-23:
You have probably figured out that you can drag both the top and the bottom handles of the layerview slider. If you drag the top and bottom handle close together, you get 1 layer. What you can do now is click the gray area between the top and bottom handle and drag that up/down.

closed 2017-07-23 by BagelOrb

PJParish, 2017-07-23:
Got it.  Now I see that the grey span is clickable and keeps the layer span.  Cleaver.  Is there a help for these learning these new hidden feature controls?

Ghostkeeper, 2017-08-03:
There is an online guide for Cura: https://ultimaker.com/en/resources/20410-view-modes

An integrated help/what's this?/tip of the day/first start guide is something we'd like to do one day, but no concrete plans have been made yet as far as I know.