Ultimaker/Cura #2147: Cut off bottom of object - Bring it back!

PJParish, 2017-07-22:
Can the Cut Off Bottom of object feature be brought back?  That was a very useful tool to have.

fieldOfView, 2017-07-23:
You can use the move tool to move a model down "into" the buildplate

closed 2017-07-23 by BagelOrb

PJParish, 2017-07-23:
I see that now.  Thanks.  Was only expecting positive movement.  Maybe a +/- note to make this more obvious?

nickthetait, 2017-07-28:
Does seem a pretty non-obvious feature.... :thinking:

Ghostkeeper, 2017-08-03:
> Maybe a +/- note to make this more obvious?

Do you mean, like, adding buttons to the move tool panel to incrementally move the object there? I can see where you're coming from, but it still doesn't feel very obvious to me how a user would interpret this as "Cut Off Bottom".

PJParish, 2017-08-03:
Maybe a ghosted down arrow on the object and/or a slider below Z that shows - at one end and + at the other with the position dot at 0 in the middle.  The span would be double the build height.  That would at least tell the user "I can move left and the object will go down".

Change the icon to show up and down with a build plate plane in the middle?

It doesn't have to be called "Cut off Bottom" just be more obvious that the model can be moved down into the build plate a controlled distance.

PJParish, 2017-08-03:
Maybe I should start a new bug thread but related to this issue the XY move coordinates don't start at 0 for X and Y where the home marker is located.  The 0,0,0, point is the middle of the bed.

PJParish, 2017-08-03:
Here is another thought.  Make the Z arrow point down by default.  Moving an object up above the bed will snap it back to flat on the bed when you release it.  Is there any reason to move it up in the first place after it is loaded on the bed?  It only stays where you put it if you move it down.

Ghostkeeper, 2017-08-03:
You can make it float by disabling the dropping down to build plate option in the preferences menu. It won't drop the model any more, but the engine still removes any empty layers at the bottom so effectively it'll still move the object down unless you have support activated. This is sometimes useful for tricky prints (or in my case for testing support generation).

PJParish, 2017-08-04:
Ah, so there is a reason to float the object above the bed and a way to do it.  Nice.

Question: Given that this issue was closed immediately after posting will any of the ideas to make it more user obvious actually get considered or implemented?

PJParish, 2017-08-05:
Rotate the icon 90 degrees and it will tell the story.

Ghostkeeper, 2017-08-06:
The thing is that the actual cutting is effectively done in the slicing process. It's very easy to just leave a few layers unprinted after the object has been sliced. We had a research project going a while ago that allowed the user to split an object through an axis exactly how the icon shows, but that hasn't made it out of the research phase yet.

Separating the "Split object" functionality off from the move tool has an additional disadvantage: What happens when the user moves the object down without splitting it? Will it crash into the build plate in an attempt to print at negative Z? Or just drop the model up towards Z=0 again? Disallow the model from being printed just like moving it outside of the build volume in any other direction?

nickthetait, 2017-08-07:
Is the research project still in progress? Has any of the information been published from this?

I do not see that as a disadvantage, merely a behavior/functional change between the two features that would need to be coordinated (made during the same release). What seems most reasonable to me is that no part of the object would be allowed to be moved below z=0.

Ghostkeeper, 2017-08-14:
The research project is put on hold since the research sprint finished and the result wasn't releasable due to a crash that occurred on some models. We have an open issue that we should fix a blocking bug in its implementation and then add it to Cura.

vrubiolo, 2019-11-17:
Glad I found this thread, it was not obvious to me how to get the old Cura option using the new UI (I went from Cura 15.x directy to 4.3!)

userosos, 2020-12-18:
Can you add full Cut an model into two parts?
If I need to duplicate the model after it i need down First model after it i need rotating  Second model and calculate for down Second model along the axis Z. This is very uncomfortable.
If you can visually move the incision area, then fully divide one model into two parts, that's ok.

sixtyfive, 2021-07-29:
As of today (Cura 4.10.0), when you move part of an object underneath the build plate, hoping to be able to print only "the tip of the iceberg", so to speak, Cura simply refuses by saying "Unable to slice" :-(

nallath, 2021-07-30:
> As of today (Cura 4.10.0), when you move part of an object underneath the build plate, hoping to be able to print only "the tip of the iceberg", so to speak, Cura simply refuses by saying "Unable to slice" :-(

In that case, please create a new issue about that (as it's a different issue!)

DougLeary, 2022-07-26:
This issue still exists with Cura 4.12.1 - if any of the part is below the build surface Cura refuses to slice it.