Ultimaker/Cura #2148: Nozzle size selection

PJParish, 2017-07-22:
Please bring the selection of the nozzle size back to the main page instead of being accessible only under the machine profiles.  Too many clicks to modify it now.

Ghostkeeper, 2017-08-03:
The intent is for you to modify the line width settings instead if you want thinner lines. The nozzle size is used somewhere deep in the bowels of Cura for merging short lines that are very close to each other, but really 99% of what it does is alter the line widths.

I guess if you switch nozzles very often (i.e. modify your printer often) then it could be useful to make the setting more accessible...

PJParish, 2017-08-04:
I can see how that makes sense.  Yes, I do switch my nozzle sizes around quite a bit.  Really easy to do.  It would be very useful to have the nozzle selection back in the configuration settings just for ease of access.    Puts line width consistent with nozzle width in the same view.

As a work around I can set up a profile for each printer (5 unique printer configs) with each of the three nozzle sizes I use and select them that way (15 choices) but it is still a lot of clicks and it makes maintaining consistency between the different profiles for a certain printer a pain as they evolve.

ianpaschal, 2018-02-27:
Since this has been inactive for more than 6 months, a workaround exists, and is quite an edge case, I'm going to go ahead and close this.

closed 2018-02-27 by ianpaschal