Ultimaker/Cura #2149: Dual head independent fan control - Enhancement request

PJParish, 2017-07-23:
Can independent control over dual fans finally be provided with speed changes at select layers in combination with the nozzle it is directed at actively printing or not?  It is a pain to modify by hand each time or manually adjust through the LCD.

Example:
Layer 1
Nozzle 1 - Printing      Nozzle 2 - Idle
Fan 1 - S000              Fan 2 - S255
Nozzle 1 - Idle            Nozzle 2 - Printing
Fan 1 - S255              Fan 2 - S000
Layer 2
Nozzle 1 - Printing      Nozzle 2 - Idle
Fan 1 - S083              Fan 2 - S255
Nozzle 1 - Idle            Nozzle 2 - Printing
Fan 1 - S255              Fan 2 - S083
...etc...
Layer 4
Both fans on at S255

Ghostkeeper, 2017-08-03:
Is there a method in g-code to communicate this to the printer?

The `M106` command has a parameter `P` that's supposed to indicate which fan to adjust the speed of. But we don't know if `P1` is directed at extruder `T1` or at some other part. Many printers have additional fans at various places, so there could be more fans than extruders.

PJParish, 2017-08-04:
Right now I have to hand code it or adjust them through the LCD interface.  I have a modified Marlin branch at the moment so i'm not sure the codes are common.  I guess it depends on the firmware.  In Marlin there are M106 P commands that are independent for two part cooling fans.  I think they are P0 and P1.  I'll confirm what they are and post the commands here.

PJParish, 2017-08-04:
Here are the M-codes that I have been using under the branch.  Looks like main Marlin uses a P instead of an F.

M106 S### will turn on both fans.  
M107 will turn off both fans.
M106 F0 S### will control the first fan only. (Primary Extruder 0)
M106 F1 S### will control the second fan only. (Second Extruder 1)
M107 F# should turn off the designated fan.

Ghostkeeper, 2017-08-21:
I wonder how consistent that is though. Cura has definitions for printers with up to 8 extruders which do nozzle swapping by parking them in a corner of the build plate, but those print heads don't have 8 fans on them.

PJParish, 2017-09-14:
That's a good point.  Some machines have a cooling fan(s) that cool all the heads at once with no independent control, and others swap heads and leave the cooling fan untouched still mounted to the carriage, while others have a separate cooling fan for each head (two heads at the moment), still others have a cooling fan for each head that gets swapped out.

Maybe what is required to make this more universal is a check box for the independent fan control option and the number of fans.  If there is only one cooling fan then it runs as coded for the active head and the parked heads are in a cooler standby mode with no part cooling fan running.  If there are two fans or two desired modes (check box checked) then the idle head and fan (and any others additional) run the same and the printing head is controlled independently as demonstrated above.  Only one head is printing at a time so all other idle print heads (2+) can be treated the same.  Still creates a two state independent control setup regardless of actual cooling fan numbers and heads.

This would potentially allow different fan control schemes based on the active head and the properties of the filament being used regardless of the actual fan configuration.

Ghostkeeper, 2017-09-20:
Still not quite sufficient.

I know mostly about Ultimaker's printers, so let me give a few examples of those:
* Ultimaker Original's dual extrusion kit has two nozzles and two fans blowing towards it. One fan per extruder. Checking the checkbox would work.
* Ultimaker 2's (unofficial) dual extrusion kit has one active nozzle and fan that get swapped out as a whole. The nozzle and fan get swapped in-place and as such there is only one fan at a time. Unchecking the checkbox would work.
* Ultimaker 3 has two nozzles and three fans blowing at them: One for each nozzle and an additional fan pointed at both. Neither option would work. I'm unsure what number the third fan would be connected to.

I can also imagine cases where the printer has more than 2 extruders but only 2 fans.

Why do you want to be able to set the fan speed of the inactive nozzle?

PJParish, 2017-09-20:
The primary reason for independent fan speed control is for additional ooze control when printing with dual heads on a single carriage when starting out the print.  First layer active head needs no cooling.  Fan is off.  The second head is hot but not printing and a fan blowing across the tip of the nozzle greatly helps with ooze.  Any that does occur does not stick as hard to the print or the shield and thus does not hang up the heads.  When the second head starts printing its first layer the fan goes off so it can stick to the bed really well and the first head fan comes on full speed to minimize ooze.  This switches back and forth till layer 5 when both come on full speed all the time.  I get very clean and precise prints this way.  At any time during the print it allows control over the active head part cooling fan while the idle head is ooze minimized by its fan.  Greatly reduces the time spent waiting for a head to heat back up from a standby temp but can be used in combination with that feature.

I appreciate your thoughtful consideration of this concept and your feedback on it.  Your examples demonstrate that it is possible to have two different states for the fans themselves.  Shared or single.

For your third situation above with two nozzles and three fans it would seem logical to drive the third fan like you do for the active head as it is affecting both heads.  Active head rules in the case of sharing a fan for two nozzles.

Only two fans for more than 2 extruders would fall under active extruder rule and sharing fans rule s well.

Provide an option to select if the fan (1, 2, 3, etc) is single or shared when independent control is selected.  This could actually provide some really nice control schemes for a lot of different configurations.

fieldOfView, 2017-09-20:
> Ultimaker Original's dual extrusion kit has two nozzles and two fans blowing towards it.

No it doesn't. It has only one fan on the head, mostly aimed at the print.

Ghostkeeper, 2017-09-25:
I'm convinced of the usefulness of separate control per fan, but I'm not convinced that it can be done with the multitude of configurations that are possible. G-code provides no guarantee which fan belongs to which nozzle, so we can't either. At least, not unless we ask a user.

We could let the user fill this in manually, but this requires a whole interface, USB/LAN connectivity to enable or disable the fans remotely, graphics to ask which nozzle the fan is blowing at... And it's all fairly complex and time consuming for a novice.

We could let the designer of the machine definition fill this in, but this requires quite a lot of settings that have very little semantic meaning, such as "is there a FAN 0" and "the nozzle that FAN 0 is blowing on" and some way to indicate that FAN 0 is blowing in the general direction of the print instead. And in the case of Cartesio those 8 nozzles being hit by 1 fan.

PJParish, 2017-09-27:
Then maybe let's start out with a simple configuration and not try to make it applicable to all possible combinations on the first go.  It can just cover most of the basic setups that people already have.  As mentioned before provide a simple check box for independent control.  If checked then use the basic Marlin codes of P0 and P1 tied to two separate nozzles T0 and T1 respectively.  Checked and unchecked will cover Ultimaker original and 2 or whatever the actual configuration is.  The third fan on the Ultimaker 3 behaves the same as the active nozzle.  For the Cartesio and similar setups the check box is off and the one fan is controlled the same as a single head. 

This will at least implement it on a simple basis in a slicer for a large majority of the basic dual head printers out there.  It seems that most of them either have a single fan blowing at both nozzles or separate fans blowing at a designated nozzle.  The Firmware of each printer will provide the pin and code assignments if they can control them independently.

All that would need to be added to the visual interface would be the check box for independent control.  This would be tied in with the height at which the fans turn on 100% all the time.  The speeds of the fans when their print head was active or idle could be set behind the scenes as described above.

Maybe I'm being too simplistic but let's start with what we can define on a few specific configuration set ups and go from there.  It will provide a valuable tool tweak for a lot of printers not found anywhere else.

Ghostkeeper, 2017-11-21:
Our project manager just removed this from Ultimaker's planning since it's more than 12 weeks old.

GregValiant, 2024-12-12:
I've added a pull request #20024 to add this function to AddCoolingProfile.  It's available for multi-extruder machines that different layer cooling fan circuits enabled.

If anyone is listening - I'll go ahead and close this as completed.

closed 2024-12-12 by GregValiant

GregValiant, 2024-12-13:
It works in all versions from 4.13.1 to current and also with other forks of Cura.  "Advanced Cooling Fan Control" has been included with UM Cura since 5.7.0.
The attached file has the changes.  Unzip it and put "AddCoolingProfile.py" it into your "Configuration Folder" and the "scripts" sub-folder.  It will be available with the other post-processors.
[AddCoolingProfile.zip](https://github.com/user-attachments/files/18125465/AddCoolingProfile.zip)

The new setting is "Enable 'Off' Speed of the Idle Fan" and is at the bottom of the list.  Enable that to access the actual speed setting box.

BTW For smaller, faster prints "By Layer" is a better choice.  "By Feature" works well for longer prints.