Ultimaker/Cura #2150: No retraction on support material - Enhancement option

PJParish, 2017-07-23:
Provide an option to turn off retraction when printing support material and/or have a different distance setting before retracting than the setting for the model.  Most of the time retraction is not needed on the support material and for high density support raises the risk of jamming and stripping the filament.

AlucardF0X, 2017-07-24:
As BagelOrb loves to say, WE DONT SUPPORT SHIT MACHINES HURRDURR (a reason he deleted most of my feature requests, which I'm sure he'll hastily forget here)

Just get a wire wheel to clean your drive gear from jamming. I got a cheap dremel wire-wheel bit and attached it to thread into the drive gear. Has worked wonders for me for a year now. No wear on the drive-gear either.

nallath, 2017-07-25:
We don't support third party machines, because we can't. We don't have time or any of those machines. 

Also, my paycheck is paid by Ultimaker. That means that they get to decide where I spend my time on. If you check around, you'd see that despite this, we've helped anyone that wanted to make changes to a non Ultimaker machine.

@AlucardF0X; If you can't hold a normal conversation, I'm going to ask you to refrain from posting here. It's fine to not agree with us, but there is absolutely no reason to get personal with us.  We delete feature requests all the time, as we get about 5-6 times the feature requests that we can actually handle.

Ghostkeeper, 2017-08-03:
This feature holds for any printer, anyway, unless the support is printed with a different extruder (in which case this is already possible since you can enable retraction per extruder).

In my experience, if you don't retract on support, you get underextrusion there which makes the support a bit wobbly. But my experience in that regard is mostly with PLA. I can see that this might be different for something with more viscosity.

Ghostkeeper, 2017-08-03:
Looking through our backlog to check before creating a ticket for this feature request, I found that we already had that feature requested. It originated from here: https://github.com/Ultimaker/Cura/issues/1828

That issue is quite different in nature though, so I'll leave this open.

PJParish, 2017-08-03:
Thank you for leaving this open.  The request is intended for single extruder machines using the same material for support and object.  Even so it would be useful on dual extruders for the same reason.  If you do it for one you might as well do it for both.

I recently had a project that required very dense support material and a high retraction rate for the small features.  The print kept failing from jamming due to retraction generated heat creep when printing the supports so I ended up making them even more dense and set the move distance before retraction to about twice the distance between the supports essentially turning it off for the supports.  The print worked with no further issues but did require a bit more clean up after.  The supports came out very strong with no underextrusion but were not that hard to remove.  I have a very good part cooling fan and have a very precise method of calibrating my filament which helps greatly but this could be applied to any printer.

By the way I always try to ask for enhancements that can be applied to any printer.  If it is for a specific printer then it is because a feature that was present before is no longer there under a particular printer configuration or I can't find the feature because it is not obvious how it has been newly implemented.

closed 2018-11-22 by Appesteijn