Ultimaker/Cura #2151: 2.6.2 Updates to duplicated profiles update parent and sibling clone settings

spinkham, 2017-07-23:
What I expect: Changes to properties like retraction distance only affect the profile they are being saved to.
What happens: Changes update the profile it was cloned from plus any clones from the same parent.

How to reproduce:
Start Cura 2.6.2
Manage Profile
Clone protected profile (I used Draft Profile)
Switch to new cloned profile
Make change to retraction settings (seems to work for other settings as well)
Save changes to cloned profile
Switch to protected profile (and/or other profile cloned from same protected profile) and see settings have changed.

Found on: Mac, cura 2.6.2, Prusia i3 printer setting. Config dir was deleted to reset to defaults when upgrading to 2.6.

nallath, 2017-07-24:
I could not reproduce this. Switching back to the protected profile set the settings to the correct value.

spinkham, 2017-07-24:
Hmm, I can't reproduce exactly either, but a can consistently a similar bug.. Not sure if I was actually switching to the profile or just checking in the managed profile screen.

Today trying on a few installs, it seems the values do look to be changed in the "profile" column of the protected profile when you look at them in the managed profile screen, but do in fact switch back the protected value when you actually switch to the new profile.

As far as I can tell, the value displayed in Profiles -> Preferences is changed by any change to clones, but the actual value applied when you switch to the profile is not.
Restarting Cura makes the manage profile view match what will be applied again.

spinkham, 2017-07-24:
Even after restart, switching to the cloned profile changes the value for the original displayed in the Profile -> Preferences screen, but the values used when switching to the original profile remain correct.

Ghostkeeper, 2017-08-06:
Couldn't reproduce the original bug either, but I can reproduce the other bug. It indeed displays my changed settings in the Profile Manager under the protected "Draft Quality" profile after performing the above steps.

Ghostkeeper, 2017-08-06:
After some further testing, I discovered that this had nothing to do with actually switching to the profile. It's just that settings that are not overridden by a particular profile are not cleared when selecting that profile in the profile manager. And so they display the value for that setting where it was last set. To demonstrate this, I created two custom profiles with the following steps:

* Add an Ultimaker 2 printer.
* Switch to Draft Quality.
* Change Retraction Extra Prime Amount to 1mm^3 (probably works with any setting).
* Save current changes as a profile: Draft Quality 2.
* Switch back to Draft Quality.
* Change Retraction Extra Prime Amount to 2mm^3.
* Save current changes as a profile: Draft Quality 3.
* Switch back to Draft Quality.
* Now go into the profile manager.
* Select Draft Quality, then Draft Quality 2, then Draft Quality 3, then Draft Quality again. The Retraction Extra Prime Amount setting displays "2" now. It should not be displaying anything.
* Select Draft Quality, then Draft Quality 3, then Draft Quality 2, then Draft Quality again. The Retraction Extra Prime Amount setting displays "3" now. It should not be displaying anything.

Ghostkeeper, 2017-11-06:
This was removed from our planning by our project manager for being older than 12 weeks.

ianpaschal, 2018-02-27:
This is fixed in version 3.2.1.

closed 2018-02-27 by ianpaschal