Ultimaker/Cura #2152: Model Placement Y coordinate Inverted

changedsoul, 2017-07-23:
Seems When you place a model on the build plate, and move it Up in Y-axis(which should be positive Y), the pos is reporting a negative Y value.

nallath, 2017-07-24:
Up is the Z axis. I just checked this with a simple model and moving Up increases the Z, right increases the X and towards the front increases the Y.

changedsoul, 2017-08-03:
Maybe I am on a diferent version than you?
[image]
You can see the Object is in the Y(+) direction, but the location shows Y(-)

I am on Cura 2.6.2

Ghostkeeper, 2017-08-06:
I think you're confused with the axis marker being in a different coordinate system. Basically, the move tool displays Cura's coordinate system, while the axis marker (in the bottom left in your picture) is the coordinate system specific to your printer. The three arrows displayed on your object indicate the positive directions for each axis in Cura's coordinate system. Cura's coordinate system has 0 in the centre of the build plate.

I agree that it's confusing. We should probably stick to one coordinate system, at least for what the user can see. Choosing Cura's coordinate system has the advantage that setting it to 0 will place the object in the centre of the build plate, always. And loading 3MF files will put the objects in a consistent place regardless of which printer you're loading it in. Choosing the printer's coordinate system has the advantage that the g-code is in the same coordinates as what you see in the interface, and that the axis marker is usually on the side of the build volume rather than in the centre, for most printers.

changedsoul, 2017-08-09:
What I find strange is I don't remember it being like this. what I mean is, I remember if the model was in the Y+ from the Bottom Left Coordinate Markers, it would also show as a positive value in the Top Left of the screens Models POS.  Maybe I am remembering it wrong, but I don't remember the models arrows being inverted in the Y axis.
Is it something I could have configured wrong in Cura? Maybe I am nuts and Its always been like this and I never paid attention to it as close as I thought?

Ghostkeeper, 2017-08-18:
There's no configuration for this.

Maybe it was different in Legacy Cura (15.04 and before)?

Ghostkeeper, 2017-11-21:
Our project manager just killed off this issue for being older than 12 weeks, so it's no longer on our planning to fix.

ianpaschal, 2018-02-27:
This is fixed in version 3.2.1.

Ghostkeeper, 2018-03-20:
No, it's not. The [0,0,0] position of a model is still in the middle of the build plate, which is inconsistent with the printer's coordinate system. When you enter a negative Y coordinate the model moves away from the printer's origin. Nothing changed in this regard; it's still confusing.

nallath, 2024-11-29:
It has been fixed for the next release. There is a preference that allows you to flip it.

closed 2024-11-29 by nallath