Ultimaker/Cura #2154: Tower is invalid

yufanyufan, 2017-07-23:
The stl file is https://www.thingiverse.com/download:3892329
Support placement setting is everywhere. 

[image]

AlucardF0X, 2017-07-24:
I've been getting a ton of these lately.
I use the experimental conical supports, and I also use support expansion. They seem to be the perpetrators of this. 

Can you share your settings? I bet you have conical support also on!

BagelOrb, 2017-07-25:
I think this is caused just by the Support Towers, which are on by default.

The problem is that they get generated even for support which is too small to even be printed.

Ghostkeeper, 2017-12-04:
Our project manager removed this issue from our planning because it's older than 12 weeks.

ianpaschal, 2018-05-08:
This no longer reproduces in 3.3.1.

Ghostkeeper, 2018-06-07:
No, it still reproduces. This is in current master:
[image]
[The project file (rename to .3mf)](https://github.com/Ultimaker/Cura/files/2080091/CheeseShaver.curaproject.zip)

closed 2023-06-24 by github-actions[bot]