Ultimaker/Cura #2155: Design same width as nozzel are not processed

damon-atkins, 2017-07-24:
Following changes to the Fine Profile:
Printer Nozzle 0.4mm
Layer Height: 0.1mm
Line Width: 0.39mm (would not work at 0.4mm)
Outer Before Inner: True (this also needed to be ticked)
Cura Version 2.6.2
[0 3D Test Print Square.zip](https://github.com/Ultimaker/Cura/files/1168711/0.3D.Test.Print.Square.zip)

AlucardF0X, 2017-07-24:
I already made an issue about this, but they closed it and deleted it.

Cura refuses to slice parts of the model that are too close to the size of the nozzle, or smaller. 
When I refer to nozzle, I also mean line width.

0.3mm walls won't be printed by 0.4mm nozzle. 
0.4mm walls won't be printed by 0.4mm nozzle.
0.41mm walls might not always be sliced by 0.4mm nozzle
0.45mm walls will be sliced by 0.4mm nozzle. 

I really need an option to allow designs with same width (or smaller) as the nozzle to be processed.

nallath, 2017-07-25:
@AlucardF0X  You mean the pull request that exactly enables this that was merged two weeks ago?
https://github.com/Ultimaker/Cura/pull/2083
https://github.com/Ultimaker/CuraEngine/pull/556
Granted, the result is not optimal, but it already improves this behavior.

damon-atkins, 2017-07-25:
Thanks @BagelOrb for looking into this.

Ghostkeeper, 2017-08-07:
I'll mark this as "Fixed" then. If you have qualms with the current implementation, we'll leave those for a new bug report.

closed 2017-08-07 by Ghostkeeper

damon-atkins, 2017-08-07:
Have you tried the attached file against the latest build?

Ghostkeeper, 2017-08-07:
[image]

To get the flat bits printed you'd have to lower the initial layer height as well, to 0.1mm or so.

damon-atkins, 2017-08-07:
Should you need too?     Same as the shell, if the wall is smaller than the shell it should still render to g-code. (effectively reducing the shell size to fit where it needs too)   Same should be for top/bottom.

One thing interesting is the 3D printer/machine definitions don't have layer minimum setting only nozzle setting..

Ghostkeeper, 2017-08-23:
Ideally you shouldn't need to, but in Cura we need to have a layer have the same layer height all over. So it's either printed with 0.27mm layer height or not at all. We've then been choosing not at all.

Also because that's easier to implement. We just take slices at certain heights instead of also having to take into account all geometry in between the slice heights.

damon-atkins, 2017-08-23:
Might be worth adding a layer height to the 3d printer setup near the nozzle size, so software can warn of any issues.   Thanks for the explanation.

I suppose, it would be slow to process at smallest layer height of the printer, and then merge layers together which are less than line width alike to the user define target layer height. i.e layer height is between printer minimum and the user target later height, rather than fixed.

Thanks for the response....