Ultimaker/Cura #2156: Feature request: Auto layer heights

ChrisRiddell, 2017-07-24:
This is something Kisslicer 1.6 beta has and slic3er in a more manual fashion.

Basically you set a range of layer heights from eg: 0.1 to .2 and when the model is sliced it will automagically set the best layer heights through out the model giving you fast printing at 0.2 in areas that don't need detail and 0.1 where the details are needed eg on sharp fillets etc.

nallath, 2017-07-25:
We've had the request before, but there are some issues with this. 

As you might know, we have a pretty big materials & processing team, who are responsible for making all the quality profiles for all materials. As far as they have been able to find out, the flow on a certain moment is quite important for the actual quality. This is why we use layer height as one of the major settings in your profile. If you simply select a profile that has layer height of 0.2 and set it to 0.1, you will get a lot less nice quality.
So if we want this to work, we would need to make interpolation profiles so that we will have settings for 0.17mm (for instance). This will then require quite a bit of testing.

ChrisRiddell, 2017-07-25:
> As you might know, we have a pretty big materials & processing team, who are responsible for making all the quality profiles for all materials.

This is something I was not fully aware of.

You have good points and something I did not really think off, maybe something to think about for a much bigger release like cura 3 or up.

closed 2018-02-05 by ChrisRiddell