Ultimaker/Cura #2157: Default material profile missing

Thijxx, 2017-07-24:
Cura 2.6.1 in Arch Linux, the default PLA profile is missing and therefore I cannot make ANY materials. I searched high and low all over the web to download the default profile but I cannot find it anywhere. Could someone please point me to the file?

Also, my material folder is empty..

`2017-07-24 17:19:01,669 - DEBUG - cura.Settings.ContainerManager.createMaterial [752]: Unable to create a new material by cloning Generic PLA, because it cannot be found for the material diameter for this machine.`

fieldOfView, 2017-07-24:
Are you running Cura from source?

The materials are in this repo: https://github.com/ultimaker/fdm_materials

Thijxx, 2017-07-24:
Yes, and thanks for the link!

closed 2017-07-24 by Thijxx

lf-, 2017-07-27:
As for the "running from source", not really, the package build script grabs a git checkout of the given tag. Seems that it misses that bit.

I submitted a bugreport and a patch to the package: https://bugs.archlinux.org/task/54948

ctag, 2017-07-30:
I agree with lf-'s comment on the bug report that the lack of materials makes for a frustrating introduction to Cura.

In the meantime [cura-fdm-materials-git](https://aur.archlinux.org/packages/cura-fdm-materials-git/) from the AUR suffices to bring default materials to the official Cura package... As far as I know; this is my first foray into Cura land.