Ultimaker/Cura #2158: Hole and External wall offset

AlucardF0X, 2017-07-24:
So, since the dawn of 3D Printing, the feature "Model Horizontal Offset" a.k.a. "Size compensation" has been a very valuable feature. 

Except I've ran into a few 3D Models that don't wanna slice correctly.
After printing, the external dimensions of the model are consistently 0.5mm smaller than designated.
The hole dimensions are spot-on. 

If I use any of the settings, it'll make the hole too big, or too small. Oof!

There really needs to be a setting that allows configuration of internal holes and external walls, rather than moving both. So I can independently resize holes (internal walls) and external walls.

A work-around is to size-compensate with 0.5mm, thus make the holes 0.5mm smaller... Then drill out the holes, but that's a pain. 

I don't think there's a slicer that currently supports this at all. So I'm a sitting duck with a few projects I'm trying to complete, just because cura decided to make external walls too small, somehow.

It isn't a "plastic shrinks" problem because I'm pretty sure across 2mm, 0.5mm is NOT plastic shrinkage, that'd be a whopping 25% shrinkage, and that's awesomely impossible. 

[image]

AlucardF0X, 2017-07-24:
It isn't shrink/overcompensation either: I guess can explain this if I'm printing it with too much compensation (-0.09mm? that's barely any), making the model thin... But then it shrinks due to cooling, and it fits the ID, at the sacrifice of the OD? 

But that's like 0.5mm for 45.5mm... that's like 0.01% shrink... But then why are the holes PERFECT as-designed 3mm? And why is the diameter CONSISTENTLY PERFECTLY shrunk 0.5mm everywhere, in diameters and lengths? 2.5mm pegs! 45mm walls, when they should be 45.5mm.. 

It isn't a weird "delete 0.5mm blyat" because my 3mm pegs got shrunk by 0.5mm to 2.5mm...

Argh I need help this is beyond me

BagelOrb, 2017-07-25:
You shouldn't think in terms of outer perimeter and holes, but in terms of concave (inward) and convex (outward) corners.

I think most modeling software makes a round hole into a regular polygon which has its vertices on the circle, which means that the edges of the polygon take short-cuts off of the model.
Perhaps Cura is even doing that when it reduces the vertex count of each layer. It has te do this because most firmware cannot handle gcode with too many verts close to each other.

But that isn't the only thing at play here. I think there are also mechanical interactions between consecutively printed line segments which cause the filament to take shortcuts when being printed.

I think the easiest way to cover several effects at once is to compensate corners outward wrt the corner.
I know there already is an extensive thread about different algorithms for this and a pull request of somebody who implemented one of the algorithms in a post-processor, but I cannot find those ATM.

BagelOrb, 2017-07-25:
Ah here it is:
https://github.com/nallath/PostProcessingPlugin/pull/47
https://github.com/Ultimaker/CuraEngine/pull/94

BagelOrb, 2017-07-25:
I think this is the most important bug for dimensional accuracy which we would have to compensate for.

AlucardF0X, 2017-07-25:
So I found a huge problem with slicing software. 

... Wouldn't it be easiest to split the "x/y horizontal offset" option into 
"x/y horizontal CONCAVE offset" and "x/y horizontal CONVEX offset" that shrink the external sides of the model and the internal sizes of the model?

Then it'd be a simple matter of trial and error to get it right. Then eventually someone will have a formula for it...

The software already has the inherent logic of identifying which direction to scale vertices towards - away from the model's VOLUME, rather than its origin (standard scale). Surely it isn't difficult to split it into two settings?

I can 99.9% fix this feature in the software, if you guys give me a guarantee that it'll get implemented, because Im too lazy to fork/pullreq and then cry in a corner when you never accept it. I'll only post the changes here in this issue report and expect you guys to implement it.

nallath, 2017-07-26:
Most of the pull requests get accepted. The few that don't get accepted is usually because we have some issues with codestyle / solution (we have a 100% peer review & test system) and the original author doesn't resolve those issues.

BagelOrb, 2017-07-26:
It wouldn't make sense to compensate the concave corner shortcuts with a
static amount of offset.
Suppose we have an almost straight wall with a few very shallow corners.
You solution idea would cause the wall to become jagged because it would be
switching between the two offsets every so often.

If you ask me then an algorithm to solve this should:
- cause less divergence from the outline for more straight corners
- don't affect long straight walls in the middle when the end is connected
to a 90° corner

AlucardF0X, 2017-07-26:
I already told you my IDs are consistently fine, my ODs are consistently 0.5mm smaller. 

If you're so afraid of jagged walls, make a setting for the detail holes are sliced down to. My printer can handle the higher amount of gcodes. 

It just feels like you guys are proactively making sure the software doesn't get updated, talking all theory without any attempt on your part to do anything. I've seen your github, you reject basically every change or let it get auto-deleted due to lack of attention, the same way you manage this issues tracker. 

I tried to contribute a suggestion several times, you guys made excuses and delete my suggestion. You do the same "appease customer with words but do no work" all over the damn forum. This is why I didn't buy a Ultimaker machine but instead a china clone. I've heared all over that Ultimaker support is the epitome of shit and this here just proves it. 

I tried to contribute, I tried to offer my skills, but you guys nallath and bagelorb are actively getting in the way of the development. I don't give a shit if you ban me or whatever. Im no longer interested in associating with cura or ultimaker, I'm making my own slicer that doesn't suck balls and won't need any updating. 

I can share with you the design since you guys are too lazy to do anything, maybe you'd patent troll it? A progressive/procedural image-based slicer that generates bitmap images, analyzes before/after layers to generate a smart toolpath on an interative bitmap process, where it manually draws with a "paint brush" head, where it'll compensate for any and all extruded plastic, no shitty "path trace" algorithms. This means that people can just input custom fill formulas similar to Photoshop's shader codes for patterns, meaning you dont need to update the software, just share slicing profiles. 

This slicer could then be used for laser cutting, or ram-EDMing. Since it would be GPU accelerated, it wouldn't suck balls like all y'all shitty slicers that can't fucking draw 100 lines on screen without crawling to a halt. And it won't be written in shitty PEETHOON FOR LEENUUXX CROSS PLATFORM INSTALL GENTOO SHITLORD languages. Have fun, I certainly didn't. Enjoy shit-talking me and climbing on your high throne.

fieldOfView, 2017-07-26:
Are you alright? How was your day?

nallath, 2017-07-27:
You know that once you share a design on teh interwebs we can't actually patent it. Noone can. 

Good luck making the slicer.

smartavionics, 2017-07-27:
Hi, I'm not an Ultimaker person and I don't use an Ultimaker printer so I can say what I like. If you work up a  PR that does what you want and it conforms to the Cura devs style guidelines and is likely to be useful to other people then it will most likely be accepted into Cura. If you produce something that is coded like shit, looks like shit and does something that only you have a use for, it probably won't get included. Of course, that doesn't stop you using a private version of Cura that includes your ideas. That's the beauty of open source, right?

BagelOrb, 2017-07-27:
Let me reiterate:
I think this is the most important bug for dimensional accuracy which we would have to compensate for.

jackha, 2018-08-16:
Closed because of inactivity.

TNTINC, 2020-01-17:
It has been almost three years now.

Will this ever be fixed?

smartavionics, 2020-01-17:
Hi folks, I've added a Hole Horizontal Expansion setting to my Cura and here's an example of its use (and limitations).

[image]

[image]

You can see that the holes that come through the sloping surface are not expanded. That's because they are not holes as they encroach on the part's outline.

Anyway, if you're interested in testing this I will be building a release over the weekend.

smartavionics, 2020-01-18:
My latest (20200118) Cura build (at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0) now has this setting. Please read the README.md there for more information on my builds.

mahtDFR, 2020-01-20:
There is a workaround by using a post processing script. If you enable 'post stretch script' you can modify the movements of the extruder to compensate for the process limitations that can make holes to small when printing.

nallath, 2020-01-21:
> Extensions > Post Processing > Modify G-Code > Add a script > Post stretch script

True, but this is extremely slow. An actual solution in the engine is to be preferred.

louleluk, 2020-01-26:
Hi guys i have test horizontal expansion with cura but for thin wall the slicer give me a lot of infill mistake!! But if anyone have tested out the streching post treatment, it will affect only the small holes or the design itself?

smartavionics, 2020-01-26:
Hello @louleluk , can you provide an example project or at least a screen shot to show the problem? Thanks.

louleluk, 2020-01-27:
finally i have tested my original design and i have enable the horiz
ontal expansion but i also have fix my problem with line width 0.3mm value
 for me the hotend have a diameter 0.4mm 
[image]here above the problem is the infill between walls.
And now below with horizontal expansion and 0.3 width line.
[image]
dimensions are correct

smartavionics, 2020-01-27:
Thanks for the info. Yes, Cura really doesn't cope well with walls that are an odd number of line widths width. Glad you have a solution.

evtrados, 2020-03-04:
This issue can now be compensated using the 'Hole Horizontal Expansion' setting, where the user can manually enlarge or shorten the holes. This setting will be available in the next release.

closed 2020-03-04 by evtrados

Paragon1970, 2020-03-21:
> This issue can now be compensated using the 'Hole Horizontal Expansion' setting, where the user can manually enlarge or shorten the holes. This setting will be available in the next release.

Great news! 
Do you have a link to the source branch that has this feature, please?

smartavionics, 2020-03-21:
See https://github.com/Ultimaker/Cura/pull/7010 and https://github.com/Ultimaker/CuraEngine/pull/1186.

Conti93, 2020-05-09:
Hole Horizontal Expansion doesn't work when the layer hasn't got an internal circle, e.g. when you have a hole on a side:
[image]
Ideas to make this fit without modifying the part?

smartavionics, 2020-05-09:
That's because on those layers, it's not a hole but part of the outline.

smartavionics, 2020-05-09:
As for a workaround, what I would do is create a cube model that is simply a wall width wide and use it to block off the horizontal tubes so that the inner region is no longer part of the model's outline. Once the part is printed, just drill through the thin walls.

smartavionics, 2020-05-09:
Not really a cube but a thin plate that is large enough to block the horizontal holes, just load the mesh and move it into position.