Ultimaker/Cura #2161: Core dump, when trying to add UM3

thopiekar, 2017-07-25:
What I find very interesting is that I'm using the builds from master and cura is still using the 2.6 config dirs. :thinking: 

```
thopiekar@UM-LP-113:/?!$ cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/thopiekar/.config/cura/2.6
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/thopiekar/.local/share/cura/2.6
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/thopiekar/.cache/cura/2.6
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [119]: Trying OpenGL context 2.0...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [128]: We got at least OpenGL context 2.0: 3.0 No profile
UM.Qt.QtApplication.__init__ [80]: Detected most suitable OpenGL context version: 2.0 No profile
2017-07-25 09:47:11,617 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-07-25 09:47:11,838 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-07-25 09:47:11,839 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-07-25 09:47:11,841 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-07-25 09:47:11,889 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-07-25 09:47:11,907 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-07-25 09:47:11,917 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-07-25 09:47:11,925 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-07-25 09:47:11,932 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-07-25 09:47:11,945 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-07-25 09:47:11,979 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-07-25 09:47:11,988 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-07-25 09:47:11,992 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-07-25 09:47:11,997 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-07-25 09:47:12,000 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-07-25 09:47:12,005 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-07-25 09:47:12,010 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-07-25 09:47:12,014 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-07-25 09:47:12,172 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin USBPrinting
2017-07-25 09:47:12,239 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice                                                           
2017-07-25 09:47:12,291 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader                                                                   
2017-07-25 09:47:12,476 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin 3MFReader                                                                            
2017-07-25 09:47:12,557 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SliceInfoPlugin                                                                      
2017-07-25 09:47:12,605 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeReader                                                                          
2017-07-25 09:47:12,642 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin AutoSave
2017-07-25 09:47:12,691 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ChangeLogPlugin
2017-07-25 09:47:12,730 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader
2017-07-25 09:47:12,765 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction
2017-07-25 09:47:12,771 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser
2017-07-25 09:47:12,777 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader
2017-07-25 09:47:12,819 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XRayView
2017-07-25 09:47:12,869 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileWriter
2017-07-25 09:47:12,870 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin CuraEngineBackend was already loaded
2017-07-25 09:47:12,935 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool
2017-07-25 09:47:12,993 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin X3DReader
2017-07-25 09:47:12,998 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView
2017-07-25 09:47:13,036 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView
2017-07-25 09:47:13,070 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter
2017-07-25 09:47:13,342 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UM3NetworkPrinting
2017-07-25 09:47:13,516 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UltimakerMachineActions
2017-07-25 09:47:13,523 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd300013d._ultimaker._tcp.local.
2017-07-25 09:47:13,524 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3002199._ultimaker._tcp.local.
2017-07-25 09:47:13,524 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000113._ultimaker._tcp.local.
2017-07-25 09:47:13,525 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd30001ab._ultimaker._tcp.local.
2017-07-25 09:47:13,525 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd30001fd._ultimaker._tcp.local.
2017-07-25 09:47:13,525 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000117._ultimaker._tcp.local.
2017-07-25 09:47:13,526 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000bdc._ultimaker._tcp.local.
2017-07-25 09:47:13,526 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000139._ultimaker._tcp.local.
2017-07-25 09:47:13,526 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000024._ultimaker._tcp.local.
2017-07-25 09:47:13,527 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd300010f._ultimaker._tcp.local.
2017-07-25 09:47:13,527 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000119._ultimaker._tcp.local.
2017-07-25 09:47:13,527 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd30001fa._ultimaker._tcp.local.
2017-07-25 09:47:13,527 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd300012b._ultimaker._tcp.local.
2017-07-25 09:47:13,528 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000162._ultimaker._tcp.local.
2017-07-25 09:47:13,528 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000167._ultimaker._tcp.local.
2017-07-25 09:47:13,528 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd300001c._ultimaker._tcp.local.
2017-07-25 09:47:13,529 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3000137._ultimaker._tcp.local.
[... cut ...]
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007f2624e48700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 63 in setItems
  File "/usr/lib/python3/dist-packages/cura/Settings/ExtrudersModel.py", line 224 in __updateExtruders
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 670 in run
  File "/usr/bin/cura", line 73 in <module>
Segmentation fault (core dumped)
```

Ghostkeeper, 2017-08-07:
Can't reproduce. Printers add fine.

> 2017-07-25 09:48:22,236 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/ttyACM0 to remove

Perhaps it has something to do with the removable drive? Or the USB printing that seems to be going on at the time of the crash? I must say that I never dared switching printers while USB printing.

thopiekar, 2017-08-10:
For me it is reproducible on my work laptop and private laptop, which are running Ubuntu and use my PPA packages.

thopiekar, 2017-08-29:
Ok, not related to the USB printing plugin. Just removed it and got almost the same:

```
thopiekar@hp:~$ cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/thopiekar/.config/cura/2.6
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/thopiekar/.local/share/cura/2.6
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/thopiekar/.cache/cura/2.6
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [119]: Trying OpenGL context 2.0...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [128]: We got at least OpenGL context 2.0: 2.1 No profile
UM.Qt.QtApplication.__init__ [82]: Detected most suitable OpenGL context version: 2.0 No profile
2017-08-29 14:27:05,094 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-08-29 14:27:05,112 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-08-29 14:27:05,113 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-08-29 14:27:05,122 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-08-29 14:27:05,123 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-08-29 14:27:05,126 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-08-29 14:27:05,135 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-08-29 14:27:05,144 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-08-29 14:27:05,151 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-08-29 14:27:05,160 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-08-29 14:27:05,168 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-08-29 14:27:05,173 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-08-29 14:27:05,177 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-08-29 14:27:05,183 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-08-29 14:27:05,186 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-08-29 14:27:05,191 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-08-29 14:27:05,195 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-08-29 14:27:05,198 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-08-29 14:27:05,202 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser
2017-08-29 14:27:05,206 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XRayView
2017-08-29 14:27:05,210 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView
2017-08-29 14:27:05,223 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UltimakerMachineActions
2017-08-29 14:27:05,240 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeReader
2017-08-29 14:27:05,244 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader
2017-08-29 14:27:05,253 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin 3MFWriter
2017-08-29 14:27:05,257 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter
2017-08-29 14:27:05,260 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileReader
2017-08-29 14:27:05,265 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XmlMaterialProfile
2017-08-29 14:27:05,269 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader
2017-08-29 14:27:05,274 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader
2017-08-29 14:27:05,340 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SliceInfoPlugin
2017-08-29 14:27:05,344 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade25to26
2017-08-29 14:27:05,348 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade21to22
2017-08-29 14:27:05,352 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade22to24
2017-08-29 14:27:05,356 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade26to27
2017-08-29 14:27:05,362 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin 3MFReader
2017-08-29 14:27:05,363 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin CuraEngineBackend was already loaded
2017-08-29 14:27:05,367 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool
2017-08-29 14:27:05,372 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin AutoSave
2017-08-29 14:27:05,378 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice
2017-08-29 14:27:05,395 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView
2017-08-29 14:27:05,414 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin X3DReader
2017-08-29 14:27:05,418 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileWriter
2017-08-29 14:27:05,423 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ChangeLogPlugin
2017-08-29 14:27:05,428 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction
[... cut ...]
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007fd59db93700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 233 in initializeEngine
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 671 in run
  File "/usr/bin/cura", line 73 in <module>
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

nallath, 2017-08-30:
This seems like a pyqt or qt version issue. I just added  an UM3 and it works as expected

thopiekar, 2017-08-30:
Ok, so I can only blame it on some ugly patches applied by Canonical and/or Debian.

Ghostkeeper, 2017-09-09:
Works fine for me on Ubuntu 17.04 as well as 16.04.3 LTS.

bzed, 2017-09-26:
@thopiekar neither debian nor ubuntu apply ugly patches. Such comments are not helpful.
The segfault happens in Debian testing, too. Somewhere deep in qt5:
```
#0  0x00007f14ff3c2d99 in QWindow::geometry (this=0x555560dcfc60) at kernel/qwindow.cpp:1633
No locals.
#1  0x00007f14ff3c87f1 in QWindow::width (this=<optimized out>) at kernel/qwindow.h:226
No locals.
#2  QWindow::qt_static_metacall (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, 
    _a=<optimized out>) at .moc/moc_qwindow.cpp:560
        _t = <optimized out>
        _v = 0x7fffffffcfe0
#3  0x00007f1500bb9bcb in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
No symbol table info available.
#4  0x00007f1500bbad40 in QV4::QObjectWrapper::getQmlProperty(QQmlContextData*, QV4::String*, QV4::QObjectWrapper::RevisionMode, bool*, bool) const () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
No symbol table info available.
#5  0x00007f1500bd205f in QV4::Runtime::method_getProperty(QV4::ExecutionEngine*, QV4::Value const&, int) () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
No symbol table info available.
#6  0x00007f14b0bd1823 in ?? ()
No symbol table info available.
#7  0x000055556201c480 in ?? ()
No symbol table info available.
#8  0x00007f14b1340388 in ?? ()
```

I need to install more debug packages to figure out where it comes from and whats going on there.

Even worse, python3-dbg segfaults earlier.

bzed, 2017-09-26:
Updating to packages from unstab;e with python3.6 resulted in a better backtrace, directly from python:

```
2017-09-26 23:06:23,219 - DEBUG - UM.Qt.QtRenderer._initialize [172]: Support for Vertex Array Objects: True
2017-09-26 23:06:23,251 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
2017-09-26 23:06:23,254 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
2017-09-26 23:06:23,254 - CRITICAL - cura.CrashHandler.show [42]:   File "/home/bzed/workspace/3d/tools/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 182, in _render
2017-09-26 23:06:23,254 - CRITICAL - cura.CrashHandler.show [42]:     renderer.beginRendering()
2017-09-26 23:06:23,254 - CRITICAL - cura.CrashHandler.show [42]:   File "/home/bzed/workspace/3d/tools/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 111, in beginRendering
2017-09-26 23:06:23,255 - CRITICAL - cura.CrashHandler.show [42]:     self._initialize()
2017-09-26 23:06:23,255 - CRITICAL - cura.CrashHandler.show [42]:   File "/home/bzed/workspace/3d/tools/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 174, in _initialize
2017-09-26 23:06:23,255 - CRITICAL - cura.CrashHandler.show [42]:     OpenGL.setInstance(OpenGL())
2017-09-26 23:06:23,255 - CRITICAL - cura.CrashHandler.show [42]:   File "/home/bzed/workspace/3d/tools/lib/python3/dist-packages/UM/View/GL/OpenGL.py", line 43, in __init__
2017-09-26 23:06:23,255 - CRITICAL - cura.CrashHandler.show [42]:     self._gl = QOpenGLContext.currentContext().versionFunctions(profile)
2017-09-26 23:06:23,256 - CRITICAL - cura.CrashHandler.show [42]: ImportError: No module named 'PyQt5._QOpenGLFunctions_2_0'
2017-09-26 23:06:23,311 - DEBUG - UM.Qt.QtRenderer._initialize [172]: Support for Vertex Array Objects: True
file:///home/bzed/workspace/3d/tools/share/cura/resources/qml/AddMachineDialog.qml:34: Error: Cannot assign [undefined] to QString
*** Error in `python3': double free or corruption (fasttop): 0x000055bea582f3d0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bfb)[0x7fdf22fccbfb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76fc6)[0x7fdf22fd2fc6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7780e)[0x7fdf22fd380e]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QMetaType7destroyEiPv+0x41)[0x7fdf1df071f1]
/usr/lib/python3/dist-packages/PyQt5/QtCore.cpython-35m-x86_64-linux-gnu.so(+0x1e318a)[0x7fdf1af4018a]
/usr/lib/python3/dist-packages/PyQt5/QtCore.cpython-35m-x86_64-linux-gnu.so(+0x1f2789)[0x7fdf1af4f789]
/usr/lib/python3/dist-packages/PyQt5/QtCore.cpython-35m-x86_64-linux-gnu.so(+0x1f2892)[0x7fdf1af4f892]
/usr/lib/python3/dist-packages/PyQt5/QtQml.cpython-35m-x86_64-linux-gnu.so(+0x7bed0)[0x7fdf1aae2ed0]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlVMEMetaObject8metaCallEP7QObjectN11QMetaObject4CallEiPPv+0x1ee)[0x7fdf1a65640e]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZNK3QV420QQmlValueTypeWrapper5writeEP7QObjecti+0xb8)[0x7fdf1a6eda48]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(+0x2c3904)[0x7fdf1a6e7904]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(+0x2c455e)[0x7fdf1a6e855e]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN11QQmlBinding6updateE6QFlagsIN16QQmlPropertyData9WriteFlagEE+0x253)[0x7fdf1a6e5323]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN12QQmlNotifier10emitNotifyEP20QQmlNotifierEndpointPPv+0x253)[0x7fdf1a6bbf63]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN8QQmlData13signalEmittedEP24QAbstractDeclarativeDataP7QObjectiPPv+0x109)[0x7fdf1a65af39]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN11QMetaObject8activateEP7QObjectiiPPv+0x9f)[0x7fdf1df1d11f]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN7QWindow12widthChangedEi+0x41)[0x7fdf18e3dda1]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(+0x10b53b)[0x7fdf18e4553b]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN11QMetaObject8activateEP7QObjectiiPPv+0x865)[0x7fdf1df1d8e5]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN7QWindow12widthChangedEi+0x41)[0x7fdf18e3dda1]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN22QGuiApplicationPrivate26processGeometryChangeEventEPN29QWindowSystemInterfacePrivate19GeometryChangeEventE+0x284)[0x7fdf18e36ec4]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN22QGuiApplicationPrivate24processWindowSystemEventEPN29QWindowSystemInterfacePrivate17WindowSystemEventE+0x22d)[0x7fdf18e3bb1d]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0xab)[0x7fdf18e13cab]
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x9c5a0)[0x7fdf0727e5a0]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2e7)[0x7fdf1cf68f67]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4b1a0)[0x7fdf1cf691a0]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c)[0x7fdf1cf6922c]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x5f)[0x7fdf1df4747f]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x13a)[0x7fdf1deece3a]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN7QDialog4execEv+0x207)[0x7fdf19821557]
/usr/lib/python3/dist-packages/PyQt5/QtWidgets.cpython-35m-x86_64-linux-gnu.so(+0x38deb7)[0x7fdf1a072eb7]
python3(PyCFunction_Call+0x4f)[0x55be99bf215f]
python3(PyEval_EvalFrameEx+0x5af)[0x55be99bda51f]
python3(PyEval_EvalFrameEx+0x423f)[0x55be99bde1af]
python3(PyEval_EvalCodeEx+0x20d)[0x55be99be35dd]
python3(+0x1c5f83)[0x55be99bf3f83]
python3(PyObject_Call+0x47)[0x55be99c342e7]
[... cut ...]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4b1a0)[0x7fdf1cf691a0]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c)[0x7fdf1cf6922c]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x5f)[0x7fdf1df4747f]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x13a)[0x7fdf1deece3a]
```

Installing python3-pyqt5.qtopengl actually fixed the segfault.
A better check for missing dependencies might be necessary.

thopiekar, 2017-10-03:
@bzed Most of the dependencies shall be detected by the debhelper tools, but still some of them seem to be detected as optional, like plugins.
Thank you for finding this package, will add it manually and force the installation of it.

thopiekar, 2017-10-03:
Ah! This is strange.. The package should be there already!
https://github.com/thopiekar/Cura-packaging/blob/master/Uranium/control#L33

thopiekar, 2017-10-03:
@bzed Which PPA are you using? Stable? Master?

bzed, 2017-10-03:
@thopiekar None, I build stuff on my own, in this case not using debhelper and friends. Just run into the exactly that issue and tried to figure out whats going on.
I'm actually not sure if the missing package was the problem at all as python should never throw a core dump because of that. I think there must be actually a strange bug somewhere in the QT stuff that triggers the coredump, maybe due to hardening compiler options.

As you are seeing the same bugs in Ubuntu: since when do you see them? Which Ubuntu are you running on your machine? Were there updates for qt5 or python3 recently?

thopiekar, 2017-10-03:
@bzed Well, before I began to work for Ultimaker I never had to add an UM3. Also during work I'm often doing other things then printing with an UM3. Therefore I have no idea for how long this issue exists.
Also I have the feeling that most of the people, who are using my PPA, are using either older Ultimaker printers then the UM3 or printers from other vendors.
Thus, I never heard of this issue from one of the users so far and it seems like I'm the only one who has got these issues at the moment.

Well, there were issues in the past, because of Ubuntu patches to Qt5, so Cura from my PPA was not working with Unity. However, Unity is dead, but no idea what else Canonical is doing to these packages. Thus I can imaging the issue comes from their patches. The vanilla Qt build inside the AppImage is not throwing any issues.

ianpaschal, 2018-05-08:
I'm going to close this since it hasn't seen any activity for 7 months. If it's still an issue let me know @thopiekar.

closed 2018-05-08 by ianpaschal

thopiekar, 2018-05-08:
Nope, that's not an issue anymore. I guess it was in the past due to
problems in Canonical's PyQt5.