Ultimaker/Cura #2165: Support for tilted print bed

ahalekelly, 2017-07-26:
There are some new printers available soon, notably the Printrbelt and Blackbelt, which have the build surface at an angle relative to the XY gantry and print head. Printing on these printers now requires a terrible workaround which doesn't work with support material, but it would be much better to have a bed tilt setting built into the slicer. Would this be possible, and what would be the best way to do this?

BagelOrb, 2017-07-26:
I would do a translation on the overhang and support polygons while they are being generated.
If you want to implement this you should have a look at CuraEngine/src/support.cpp. Every time where polygons of different layers are compared a translation should be applied.

If you do implement this be sure to make a pull request in order to get it merged into Cura, so that the code will be kept up to date.

The frontend would need a lot of rework, though - if you want them to be properly supported.
You need to choose whether you would want to tilt the build plate or tilt other stuff, e.g. you'd have to tilt the models before sending them to CuraEngine. Currently the assumption that the build direction is equal to the direction of gravity and perpendicular to the build plate is made everywhere. It's probably conceptually difficult to separate which code depends on which of those variables.

I suppose the firmware reinterprets Z movement as a combination of Z and X movement, cause otherwise it would be impossible to use current slicers for those printers.

I hope this helps!

BagelOrb, 2017-07-26:
There's all sorts of features in the engine which are meant for bed adhesion which don't make sense any more. Once the basics have been implemented (like support) it would be time to implement specialized features for belt adhesion.

Also: for support you need to intersect with the build volume square. Otherwise the support would be generated to below the belt.

Ghostkeeper, 2017-08-07:
The front-end work involves rotating the build volume and then making sure that models load un-rotated. Also, changing the angle in which we draw the overhang areas red in the shader. These are fairly straightforward.

I think most of the work is in CuraEngine. You'd want to draw extra walls in the place where the model/support hits the belt, by way of bed adhesion. You wouldn't want a support infill pattern to lean only on one endpoint of a line.

A lot of features indeed don't make sense with a sloped build plate: Initial layer height, brim, raft, skirt, initial layer anything actually, the fan speed control we currently have, etc... There's also stuff that would have to be radically changed, such as draft shields, prime towers, ooze shields, support in some way, etc.

ianpaschal, 2018-03-02:
> Here at Ultimaker we have no belt printer, so we don't really have a means to test anything. As such I don't see this change happening any time soon.

I think it's safe to indefinitely close this for now. If the maker of such printer wants to make a PR with the required changes it might be _possible_ to integrate, but otherwise I'd say this is just not going to happen.

closed 2018-03-02 by ianpaschal

fieldOfView, 2018-03-02:
See http://github.com/blackbelt3d/cura

ianpaschal, 2018-03-02:
@fieldOfView Cool! Thanks for linking.

fieldOfView, 2018-03-02:
The bb-3.2 branch moves everything into a plugin, so a future version could be available through the plugin browser.