Ultimaker/Cura #2167: If printer is not on when Cura is started cura will not connect

astecme, 2017-07-26:
If a printer is not turned on when Cura is started Cura will not connect. The only way to connect afterwards is to shut Cura down and restrt it.

Ghostkeeper, 2017-08-07:
I assume you're talking about a USB cable, considering your previous bug reports.

I can't reproduce this problem. I can start Cura, then connect a printer via a USB cable, and Cura will detect the new device and connect to it. Perhaps Cura tries to connect to some other USB port on your computer? Cura can only handle one USB connection at a time, after all...

astecme, 2017-08-07:
NO, its as I said.

Ghostkeeper, 2017-08-07:
I believe you! But for us to fix the problem we need to reproduce the problem on our end as well. How can we reproduce it? What operating system are you using, perhaps? Or maybe the printer model you're connecting to?

closed 2018-03-02 by ChrisTerBeke