Ultimaker/Cura #2172: Nozzle size smaller than 0.3mm not supported (cura 2.6.2)

roywillow, 2017-07-27:
I tried to add my custom FDM printer with a 0.2mm nozzle, but the nozzle size was forced to set to 0.3mm. Any value smaller than 0.3 will be change to 0.3mm. Hope to fix this.

fieldOfView, 2017-07-27:
I can't reproduce this. This is what I tried:
* open Cura
* add a new Custom FDM Printer
* in the machine settings dialog that pops up, enter 0.2 for nozzle diameter
* close dialog
* go to "Manager printers..."
* open "Machine Settings"

result: nozzle size is correctly still at 0.2 mm

Ghostkeeper, 2017-08-08:
I think I found the bug he may be referring to. This is what I did:

* Make the line width setting visible.
* Add a new Custom FDM Printer.
* Set the number of extruders to 2.
* In the first extruder, change the nozzle size to 0.2mm.
* Close the dialogue.

Now the line width setting still says 0.4mm. When you switch the tabs (or otherwise update the setting display) it properly displays a line width of 0.2mm. This seems to be an update bug.

Was that what you were seeing, Roywillow?

Ghostkeeper, 2017-11-06:
This was closed by our project manager for being older than 12 weeks.

nickthetait, 2017-11-08:
@roywillow were you able to reproduce this issue? Anyone else out there experience this problem?

ianpaschal, 2018-07-30:
This is fixed in Cura 3.4.1.

closed 2018-07-30 by ianpaschal