Ultimaker/Cura #2173: Restart Cura for every STL File

DanielQuintana2, 2017-07-27:
When using Cura for multiple 3D prints I have to restart it to upload to an SD card. For example, I open cura, slice an STL, upload to SD. Then if I want to print another in the same session it wont let me upload to an SD. I found that closing it then reopening it works but it takes a lot of time.

patricknn, 2017-07-29:
This is also happening to me since the update to 2.6.2
Every time that you upload to SD the Cura freezes in a way that you can't do it again.

Vandrasc, 2017-07-31:
It should be fixed in the next release.

MFHFozzy, 2017-08-01:
And when *is* the next release?

Ghostkeeper, 2017-08-14:
Next week, most likely.

Ghostkeeper, 2017-10-03:
I can't verify that since the bug didn't occur in Linux. Could you try, @ChrisTerBeke?

ChrisTerBeke, 2017-10-03:
I do not experience this issue anymore when using the Save to file (or SD) export multiple times in a row with different models without restarting Cura (on Windows 10, 3.0 branch). I also didn't get this on 2.7 though (was on Mac back then).

closed 2017-10-12 by Ghostkeeper