Ultimaker/Cura #2178: Version 2.6.2

PJParish, 2017-07-29:
When selecting Rigidbot or Rigidbot Big as the printer there is no longer an option under printer machine settings for Dual head configurations.  I can see Dual extrusion in the setting visibility list but there are no settings under it.

PJParish, 2017-07-29:
There is also no X and Y offset for the second head but that may show up when the options for Dual head are turned on.

fieldOfView, 2017-07-30:
The printer definition has to support multiple extruders, ie there need to be json files to describe the extruders. Try setting up your printer as a "Custom FDM Printer"

PJParish, 2017-08-02:
Yes, they existed for the printer in the previous version.  I can play around with creating a custom profile and see if I can get that to work.  Thanks for the feedback.  It would be nice to have it back in the printer definition like it was before.

Ghostkeeper, 2017-08-14:
Looking through the history for these files reveals that there were never any extruder definitions for the RigidBots: https://github.com/Ultimaker/Cura/commits/master/resources/definitions/rigidbot_big.def.json
It hasn't ever been multi-extrusion, at least not using factory-default Cura. Perhaps you had a customised definition that you found somewhere?

Ghostkeeper, 2017-08-14:
If you need dual-extrusion support for the RigidBot printers, you're going to have to provide the setting files (.def.json) yourself, ask another RigidBot user to do that, or use the Custom FDM Printer. The developers of Cura can't add the definition files, since we don't have the printer available to us to test with.

PJParish, 2017-08-14:
??? I think we are not talking about the same Cura version....  I have been using version 15.04.x for quite some time to successfully slice for my dual head Rigidbots and now ReliaBuild 3D printers as a straight download from the Cura site.  I select "Other" and Rigidbot or Rigidbot Big and go into the machine settings and select 2 heads.  All the dual print functions are there.  I did nothing special.  Is there something that can be transferred from that version to the new 2.7.x one?  The .def.json file used before will not have changed.

Ghostkeeper, 2017-08-15:
Ah, no I don't think we are talking about the same Cura version then.

> Yes, they existed for the printer in the previous version.

By "the previous version" I hear Cura 2.5!

> Is there something that can be transferred from that version to the new 2.7.x one? The .def.json file used before will not have changed.

The build volume size and nozzle offsets can be transferred from Daid's Legacy Cura to Cura 2 by an expert (but again would have to be tested). There are significant changes in the slicing technique of Cura in the last 3 years, so much that I won't expect those files from December 2014 to translate well. But we can take over the nozzle offset and such physical printer properties.

closed 2018-03-02 by ChrisTerBeke