Ultimaker/Cura #2181: Better overhang quality

zerspaner-gerd, 2017-07-30:
I get better overhang quality by printing the "Top / Botton Pattern" with Concentric?

Since the inner "Top / Botton Pattern" layers are printed in Concentric, maybe you can add a setting that only prints the "Concentric" overhangs

[image]
[image]

But only on the side of the overhang, at bridges, it has a negative effect.

Here is a picture of my test prints, and this confirms my impression
[image]

Test file:
https://www.youmagine.com/designs/make-2015-3d-printer-shoot-out-test-geometries - part Overhang Test

Ghostkeeper, 2017-08-14:
Concentric bottom (tee hee) produces better bridging, it's well known.

The problem with the concentric pattern on the top/bottom is that it produces a fairly extreme blob in the middle where the concentric lines centre around. It also has some bad worst-case properties for how the pattern lies on top of the infill.

We'll have bridging areas eventually. Perhaps then we'll be able to modify the line directions with more fidelity like this.

zerspaner-gerd, 2017-08-27:
> The problem with the concentric pattern on the top/bottom is that it produces a fairly extreme blob in the middle where the concentric lines centre around. It also has some bad worst-case properties for how the pattern lies on top of the infill.

The problem I have also observed, so it would be nice if there was a setting only for the overhang areas.

The Extra Skin Wall Count feature does something similar.

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-26:
We expect to improve this via better bridging detection rather than via skin patterns on overhang parts. Discussion is ongoing.