Ultimaker/Cura #2182: Cura 2.6.2 not working with custom materials.

MFHFozzy, 2017-07-30:
Whether creating, duplicating, or importing.
Create custom material. New material shows up in list. New material file is in config folder.
Close Cura, reopen.
New material is NOT in list.
Import, select material file, get window that new material has been imported *successfully*
New material is NOT in list.

Windows 10, fully updated.
New material from file that was created by Cura:
<metadata>
    <name>
      <brand>Hatchbox</brand>
      <material>PLA</material>
      <color>White</color>
      <label>Hatchbox PLA 1.75mm 1.27g_cm3 White</label>
    </name>
    <color_code>#ffffff</color_code>
    <adhesion_info>Print on bare glass. Use tape for cold build plates.</adhesion_info>
    <GUID>5ae4606f-e668-453c-8923-19a90f4b9ff8</GUID>
    <version>1</version>
    <compatible>True</compatible>
    <description>Fast, safe and reliable printing. PLA is ideal for the fast and reliable printing of parts and prototypes with a great surface quality.</description>
    <setting_version>1</setting_version>
  </metadata>
<properties>
    <diameter>1.75</diameter>
    <density>1.27</density>
  </properties>
  <settings>
    <setting key="standby temperature">175</setting>
    <setting key="heated bed temperature">60</setting>
    <setting key="print temperature">200</setting>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2 Extended+" />
      <hotend id="0.25 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.8 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.6 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.4 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 3" />
      <setting key="print cooling">100</setting>
      <hotend id="AA 0.8">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="BB 0.8">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="BB 0.4">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="AA 0.4">
        <setting key="hardware compatible">yes</setting>
      </hotend>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker Original" />
      <setting key="standby temperature">150</setting>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2 Extended" />
      <setting key="standby temperature">150</setting>
      <setting key="processing temperature graph">
                </setting>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2+" />
      <hotend id="0.6 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.8 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.4 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.25 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
    </machine>
    <machine>
      <machine_identifier manufacturer="IMADE3D" product="IMADE3D JellyBOX" />
      <setting key="heated bed temperature">55</setting>
      <setting key="print temperature">210</setting>
      <hotend id="0.4 mm 2-fans">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="0.4 mm">
        <setting key="hardware compatible">yes</setting>
      </hotend>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 3 Extended" />
      <setting key="print cooling">100</setting>
      <hotend id="AA 0.4">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="BB 0.4">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="BB 0.8">
        <setting key="hardware compatible">yes</setting>
      </hotend>
      <hotend id="AA 0.8">
        <setting key="hardware compatible">yes</setting>
      </hotend>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2" />
      <setting key="standby temperature">150</setting>
      <setting key="processing temperature graph">
                </setting>
    </machine>
    <machine>
      <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2 Go" />
      <setting key="standby temperature">150</setting>
      <setting key="processing temperature graph">
                </setting>
    </machine>
  </settings>
</fdmmaterial>

Ghostkeeper, 2017-08-14:
Pasting that XML code directly into Github doesn't seem to go over well...

The materials disappearing after restarting Cura should be fixed for 2.7. Perhaps you can try again in the beta or when the 2.7 stable is released next week.

When importing a material, make sure that it's supported for your current set-up (i.e. current printer, current printer modifications and current printer variant). Seems to work fine when I try it.

MFHFozzy, 2017-09-06:
Testing on 2.7 release:
Its *kinda* working now?

I can get the material to save...until i set the Diameter to 1.75mm. 
I get a message now when i leave the diameter edit box.
"The selected material diameter causes the material to become incompatible with the current printer".
Upon getting that message, the material is removed from the list.
Any diameter  less than 2.5mm causes this. After this message, the material is always gone after close/reopen program:

I made a new material and never made the diameter "invalid", closed cura, reopened, material was there.

Made new material, changed the diameter to something invalid, got the warning, saw the material disapear, changed the diameter again (to a valid number), saw the material re-apear, closed cura, re-opened, and the material was gone again.

Another "small" issue with materials:
When picking the color of a custom material by clicking the color box, after picking a color, the color of the box is not changed. Gets kinda confusing.

Ghostkeeper, 2017-09-11:
> When picking the color of a custom material by clicking the color box, after picking a color, the color of the box is not changed. Gets kinda confusing.

I could reproduce this. That's indeed a bug.

Ghostkeeper, 2017-09-11:
> Made new material, changed the diameter to something invalid, got the warning, saw the material disapear, changed the diameter again (to a valid number), saw the material re-apear, closed cura, re-opened, and the material was gone again.

This one doesn't reproduce for me. The material is visible after restarting Cura, just like before restarting Cura.

ianpaschal, 2018-05-22:
>>When picking the color of a custom material by clicking the color box, after picking a color, the color of the box is not changed. Gets kinda confusing.

>I could reproduce this. That's indeed a bug.

Fixed in 3.3.

>>Made new material, changed the diameter to something invalid, got the warning, saw the material disapear, changed the diameter again (to a valid number), saw the material re-apear, closed cura, re-opened, and the material was gone again.

>This one doesn't reproduce for me. The material is visible after restarting Cura, just like before restarting Cura.

Also doesn't reproduce for me, and given the age of the ticket, I think this is safe to close.

closed 2018-05-22 by ianpaschal