Ultimaker/Cura #2183: Config for rapide lite 200

albfan, 2017-07-31:
Any chances to add a config for rapide printers

http://www.rapide-3d.com/

I can test whatever config

Ghostkeeper, 2017-08-14:
We won't make them ourselves, since we don't have a way to test what we create without having access to the physical printer. We won't know which settings need to be tweaked. People with the printer in questions are free to provide a printer definition to be added to Cura.

albfan, 2017-08-15:
Last time I achieve to print correctly was with cura 15 (which have a lot more options to tweak)

http://www.print3dforum.com/showthread.php/15-Cura-Settings-for-Rapide-Lite-200

Where all those config went?

I can provide config as PR, but only if I see it printing correctly again. Now is like creating a ball of wool

Ghostkeeper, 2017-08-29:
We've rewritten large parts of Cura for version 2.0 (since june 2015). Among other things, the format for printer profiles had changed dramatically so we decided to not take over the old profiles. They wouldn't be applicable or working correctly any more anyway. It would create a ball of wool.

All of the original options of Cura 15.04 are still there, though some have been renamed since then. There's also about 300 new settings.

albfan, 2017-08-29:
Hope be able to manage to rebump it

closed 2018-03-02 by ChrisTerBeke