Ultimaker/Cura #2185: Save project filename has double 'curaproject.3mf'

NigelWebbZUK, 2017-07-31:
Cura 2.6.2, Windows 7,
Start a new project & open a stl file, as example foobar.stl
Click File > Save Project
Save Project Summary appears > Click Save
Save to File dialogue appears with the file name UM24_foobar.curaproject.3mf.curaproject.3mf
1. Why a double 'curaproject.3mf'? There should be only one .curaproject.3mf
2. Additionally can there be an option to remove the automatic 'UM2x' from the start of the file name. When doing a directory listing they appear in the wrong place. You could add '_UM2' to the end of the file name instead and that would avoid the directory list becoming fragmented.
Cheers
Nigel

smartavionics, 2017-07-31:
Don't know about 1 but as for 2, you can disable the prefix by unchecking the "add machine prefix to job name" checkbutton in Cura preferences (been an option for quite a while now).

Ghostkeeper, 2017-08-15:
The double file extension is something we found as well and should be fixed for version 2.7.

ChrisTerBeke, 2017-09-28:
Don't see this issue anymore in 2.7 or 3.0/master. Closing for now.

NigelWebbZUK, 2017-11-08:
The double 'curaproject.3mf' is not fixed in Cura 3.0.4 (Windows 7)
Using a brand new file name _MyFileName_ that has never been saved before
Selected: File > Save Project
In the 'Save to File' dialogue box the File Name appears as: '_MyFileName_.curaproject.3mf'
When I click Save the file is saved as: _MyFileName_.curaproject.3mf.curaproject.3mf

I have found a work-around. 
In the 'Save to File' dialogue box, amend the file name by deleting the '.curaproject.3mf'. 
After the save the file correctly has only one 'curaproject.3mf'

Note, sometimes the Save As dialogue appears as _MyFileName_.curaproject.3mf.curaproject.3mf. When I save this file it saves as _MyFileName_.curaproject.3mf.curaproject.3mf.curaproject.3mf

Ghostkeeper, 2017-11-14:
:confused: It's working fine here on Windows 8.1. The file dialogue says `UM2E_Cube1cm.curaproject.3mf` and that's also how it's saved.

NigelWebbZUK, 2017-11-15:
Windows 7 SP1 & Cura 3.0.4
Two processes below, one saves correctly, the other not!
Using the Desktop as a path to save to.

**1. Saves correctly**
Open Cura
Do not load a STL file (no part loaded, blank build platform)
File > Save Project
Dialogue box 'Summary - Cura Project' click Save
'Save to File' dialogue box appears with :
-Blank 'File name:' 
-'Save as type' 'Cura Project 3MF file (*.curaproject.3mf)'. 
Type in a file name 'Foo' and click save. 
File is saved as '/Foo.curaproject.3mf' - This is correct, as it should be.

**2. Does not save correctly!**
Open Cura
Load a STL file called 'Part1.stl' (this happens to be a cube about 10mm square)
File > Save Project
Dialogue box 'Summary - Cura Project' click Save
'Save to File' dialogue box appears with :
-'File name:' 'Part1.curaproject.3mf.curaproject.3mf' 
-'Save as type' 'Cura Project 3MF file (*.curaproject.3mf)'. 
Click save. 
File is saved as 'Part1.curaproject.3mf.curaproject.3mf.curaproject.3mf' - This is NOT correct.

Ghostkeeper, 2017-11-24:
Can someone reproduce this on Windows 7? I have no computer available with that version.

ChrisTerBeke, 2018-03-02:
Can't reproduce. Closing until more info is provided.

closed 2018-03-02 by ChrisTerBeke