Ultimaker/Cura #2186: Instructions to build a tagged version

albfan, 2017-07-31:
I'm trying to build a stable version, but cannot find a way to do it.

I go into tags and all say CURA_VERSION=master.

I take a loot into cura-build repo but don't see how to specify a tag

Ghostkeeper, 2017-08-15:
There's CMake cached variables for that in cura-build. You should be able to see them in CMake-GUI or ccmake.

The variables are called:
* `CURA_VERSION_MAJOR` (currently 2)
* `CURA_VERSION_MINOR` (currently 7 on the beta)
* `CURA_VERSION_PATCH` (currently 0 on the beta)
* `CURA_VERSION_EXTRA` (currently "BETA" on the beta)

Ghostkeeper, 2017-08-15:
Oh, wait, you meant a tag where it should pull from. That property is called `TAG_OR_BRANCH`, also in the CMake cache of the cura-build repository.

albfan, 2017-08-15:
Will give it a try.

I'm trying to rebuild an old version of cura which normally drive you into an impossible dependency hell for outdated deps.

Let's see how far I get

ianpaschal, 2018-02-27:
Since we did not hear back for 6+ months, I will consider this question answered for the time being.

closed 2018-02-27 by ianpaschal

albfan, 2018-02-27:
Sure. I'm thinking on create a flatpaked cura, but didn't get any progress yet. If I manage to do it I will post here.

Till them my 3d printer is a big paperweight 😢

fieldOfView, 2018-02-27:
Why can't you use the appimage instead of a flatpak?

albfan, 2018-02-27:
@fieldOfView I'm agnostic here, I just don't know appimage, any good tutorial around?

fieldOfView, 2018-02-27:
You don't have to build an appimage, one is available from https://software.ultimaker.com
General instructions to build are here: https://github.com/Ultimaker/cura-build

albfan, 2018-02-28:
@fieldOfView I want to download 15.04.6 but is a .deb package not a AppImage

fieldOfView, 2018-02-28:
15.04 is three years old and not supported. See http://github.com/daid/legacy-cura for it.