Ultimaker/Cura #2188: Cura 2.6.2 - Support interface roof and floor thickness used even if interface is disabled

ClockworkFauxx, 2017-08-01:
Sunder supports, if the options Enable Support Interface, Support Roof Thickness, and Support Floor Thickness are visible, and the values for Support Roof Thickness and Support Floor Thickness are set, they are used regardless of whether the Enable Support Interface checkbox is ticked.

Ghostkeeper, 2017-08-02:
I'm not able to reproduce this bug. What I did was:

* Make every setting visible.
* Enable support.
* Enable support interface. The Support Roof Thickness and Support Floor Thickness are now visible.
* Change the support roof thickness to 6mm.
* Disable support interface. The Support Roof Thickness and Support Floor Thickness settings now disappear.

If I then let it slice there is no support interface in the model. The thickness settings don't get used at all.

Did you do something else? Perhaps something involving multiple extruders?

ClockworkFauxx, 2017-08-02:
I only have single extruder machines – I will attempt to reproduce it again – I had interface lasyers turned on, manually specified roof and floor values, then I loaded a different model and turned off interface layers, but they still were present in the gcode/preview until I set the thickness to 0. The settings dialogs now require more checkboxes than previous to utilize these functions – the profile I used was upgraded from 2.5 through 2.6.1 to 2.6.2 – it may have been due to that.

Ghostkeeper, 2017-08-15:
Still doesn't reproduce for me on 2.7 using these steps:
* Load a model that has both support roof and bottom (in my case the letter C).
* Enable support.
* Enable support interface.
* Set the support roof height to 2mm.
* Set the support floor height to 2mm.
* Let it slice. The support interface now appears and is indeed 2mm thick.
* Disable support interface. Let it slice.
The support interface is gone now, as expected.

closed 2018-03-02 by ChrisTerBeke