Ultimaker/Cura #2189: Cura 2.6.1 + Ubuntu 16.04LTS Printer Not Connected

jpathuber, 2017-08-01:
Apologies in advance as I'm new to Cura. BTW, works great but I have to copy and physically insert the SD card in the printer (MP Select III mini) as I can't figure out how to force the software to recognize the tty port. Also I've tried this on two Ubuntu computers w/16.04LTS with the same results. Both systems recognize the printer, the lines in /var/log/kern.log (slight variance between them, not significant):
usb 3-1: new full-speed USB device number 2 using xhci_hcd
usb 3-1: New USB device found, idVendor=cfce, idProduct=0300
usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 3-1: Product: Malyan 3D Printer
usb 3-1: Manufacturer: Malyan System
usb 3-1: SerialNumber: 206F36624150
usb 3-1: ep 0x83 - rounding interval to 1024 microframes, ep desc says 2040 microframes
cdc_acm 3-1:1.0: ttyACM0: USB ACM device
usbcore: registered new interface driver cdc_acm
cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

So the OS sees a serial port ttyACM0: but Cura 2.6.1 can't see it. AFAIK, I have exhausted all settings in the software. Any suggestions?

Ghostkeeper, 2017-08-15:
Yes there aren't many settings that adjust this in the interface.

In 2.7 we switched to a better method of detecting USB devices. It fixed this problem for others, but stuff like this depends a lot on the device and the operating system. I hope it fixed the problem for you as well. Could you try this in the 2.7 beta? Or perhaps wait for the 2.7 stable?

LRonHubs, 2017-10-19:
Uhh could you please explain that method?

fieldOfView, 2017-10-19:
See https://github.com/Ultimaker/Cura/pull/2107 for details.
We're now using the functionality built in to pySerial to list serial ports, instead of rolling our own serial port detection code.

closed 2018-03-02 by ChrisTerBeke