Ultimaker/Cura #2190: Cura 2.7 - beta crash on quit

nicolinux, 2017-08-02:
Every time I quit Cura, I get the standard Mac crash dialog "Cura quit unexpectedly".
macOS Sierra 10.12.6

Vandrasc, 2017-08-02:
As a workaround, you can close Cura using the [X] option instead of Quit Cura or Command Q. It is under our radar.

nicolinux, 2017-08-02:
It is still a bit confusing though. The convention on macOS is to close windows with the [x] and completely quit an application with cmd+q. I would expect Cura to remain open in the background when I close the window - like Chrome, Mail and like most applications.

rtideas, 2017-08-11:
I have the same issue. I have for years, pressed CMD-Q to quit applications. Cura crashes every time.

Just tried using the red dot close button and it worked without crashing--BUT I got a crash when I tried to reopen. 

Nice to have a workaround, but not really the best route. Most applications I use, the red dot closes only the window not the application.

Vandrasc, 2017-08-16:
Fixed in 2.7

closed 2017-08-18 by Ghostkeeper