Ultimaker/Cura #2191: Cura 2.7 beta - filament costs and weight not saved

nicolinux, 2017-08-02:
When I add the cost to a standard material it is reset to 0 when I restart Cura.
macOS Sierra 10.12.6

[image]

nickthetait, 2017-08-02:
This problem also exists on Ubuntu 17.04
Both fields are reset after closing Cura.

Ghostkeeper, 2017-08-18:
It doesn't happen for me. I also tried on Ubuntu 17.04. These were my reproduction steps:

* Start a fresh install of Cura.
* Add Ultimaker 3 printer.
* Go into the material manager.
* Set Filament Cost to 1 euro and weight to 200g.
* Close the dialogue.
* Restart Cura.
* Open the material manager again.

The information tab still shows 1 euro and 200g then, for me.

nickthetait, 2017-08-18:
I suggest trying with a installed version of the application @Ghostkeeper

Ghostkeeper, 2017-08-29:
Okay, I'm trying with a fresh install (or rather, deleted `AppData/Local/cura` and `AppData/Roaming/cura`) of the Cura 2.7 beta on Windows 8 now:

* Start a fresh install of Cura.
* Add Ultimaker 3 printer.
* Go into the material manager.
* Set Filament Cost to 1 euro and weight to 200g.
* Close the dialogue.
* Restart Cura.
* Open the material manager again.

The information tab still shows 1 euro and 200g then, for me :confused:

nickthetait, 2017-09-01:
Tested on Windows 10 and cannot reproduce the issue. Seems specific to Linux/OS X?

ianpaschal, 2018-02-27:
This is fixed in the most recent version (3.2.1).

closed 2018-02-27 by ianpaschal