Ultimaker/Cura #2194: Cura 2.7 beta - file name does not include printer prefix

nicolinux, 2017-08-02:
Even though the setting "Add machine prefix to file name" is activated, Cura does not add it.
macOS Sierra 10.12.6

[image]

fieldOfView, 2017-08-02:
It uses a different name altogether (shell vs insert). Not sure what happened here. Steps to reproduce?

nickthetait, 2017-08-02:
Cannot reproduce this using Ubuntu 17.04. Machine prefix is correctly added.

Ghostkeeper, 2017-08-18:
The prefix is correctly added for me too. When I change the name after it has loaded and then save, the new (changed) name is correctly saved for me.

ianpaschal, 2018-02-28:
6+ months old and not possible to reproduce. Also pretty sure we've made fixes/improvements in the last few weeks that would have corrected this if it was still going on.

closed 2018-02-28 by ianpaschal