Ultimaker/Cura #2197: CURA 2.7 [idea]

darkdvd, 2017-08-02:
Some shadow between menus and the background in the "dark" theme will be great !

darkdvd, 2017-08-02:
[image]

Ghostkeeper, 2017-08-18:
Indeed, and it would have to be themeable.

Ghostkeeper, 2017-11-17:
Our project manager just removed this from our planning because it's older than 12 weeks.

ChrisTerBeke, 2017-11-17:
There's a new design since 3.0 in which the dark theme menu's are fixed as well.

Ghostkeeper, 2017-11-17:
What do you mean by "fixed"?

ChrisTerBeke, 2017-11-17:
Personally I'm not a fan of drop shadows. The issue is that there's not enough contrast between sidebar/topbar/toolbar and 3D viewer background. Maybe toning down the background color a bit will help for the contract without needing drop shadows. We don't use drop shadows anywhere else in the design of Cura.

darkdvd, 2017-11-17:
Come on, man! It's time to make CURA sexy!

ianpaschal, 2018-05-23:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-23 by ianpaschal

darkdvd, 2018-05-23:
:-(

ianpaschal, 2018-05-23:
Cura 4.0 is on the radar and will likely involve lots of UI reworks (including, I hope, a better theming system). In light of that we're not going to invest much time in modifying the UI unless it's to accommodate bug fixes or new features until then. Sorry!

ChrisTerBeke, 2018-05-23:
Please use QtQuick2 with reusable components :)