Ultimaker/Cura #2198: Cura 2.7.0  Visability  Issues

lightening42, 2017-08-02:
[image]
Several visibility issues:
1) The view mode popup block seeing the complete model  and in not moveable.
2) When hovering over the Print symbol at the top of the screen something is popping up but cannot net see it.
3) The right side panel is too big, too much white space,  Would be nice to turn it off or slide to the right off screen and made the model bigger.

nickthetait, 2017-08-02:
1. For shorter screens this is a pretty serious problem. The view mode popup cannot be hidden or moved. I would expect to be able to click the view mode button to hide it.

2. I cannot reproduce this on Ubuntu 17.04, no indication of a popup is show. Could you try getting a screen-shot of this?

3. There are certainly situations where it would be nice to shrink the right panel.

Ghostkeeper, 2017-08-18:
1. Our User Experience team agrees. With 3.0 we've moved that view mode button to the top-right of the 3D screen to alleviate the problem a bit, but it's a hard problem to solve completely.
2. Nothing should be popping up.
3. The right panel is currently too small for Custom mode and too large for Recommended mode. We have an open issue to make it scaleable.

fieldOfView, 2017-08-18:
>  With 3.0

interesting

nallath, 2017-08-18:
Don't get too exited ;)

ianpaschal, 2018-02-28:
This is a non-issue in newer versions.

closed 2018-02-28 by ianpaschal