Ultimaker/Cura #2202: 2.7 Beta cannot switch to dark theme (crash)

filipgoc, 2017-08-03:
Mac OS Sierra
Cura Beta 2.7

### Reproduce problem: 

1. go to settings and switch to dark theme
1. close the settings
1. close cura
1. cura crashes on close
1. start cura
1. theme is still the default white one

cheerio!

biocode3D, 2017-08-03:
Same thing on Linux.
Dark theme won't take If I "quit" from the menu. I also get a segfault when it exits.

But it works if I kill the window! Theme change takes, no segfault.

fieldOfView, 2017-08-04:
Note that this is not strictly limited to setting the dark theme, but it is true for any preference. As a workaround, either wait ~10 seconds before closing Cura (so autosave kicks in) or close Cura by closing the window instead of using Quit in the menu

filipgoc, 2017-08-04:
@fieldOfView good workaround with the 10s. 

however, for me Cura crashes even when closing the window. (i.e., not using the quit function.)

Ghostkeeper, 2017-08-20:
The crash when closing the window via the X button on MacOS has been fixed now.

closed 2017-08-21 by nallath