Ultimaker/Cura #2203: SOLIDWORKS plugin fails to load models.

gabe1294, 2017-08-03:
I am unable to load SOLIDWORKS part or assembly files using the Plugin in 2.7 Beta

Vandrasc, 2017-08-04:
I can't reproduce with the following steps:
- Plugin Browser -> Download SolidWorks plugin
- Restart Cura
- Drag'n drop SolidWorks file into Cura (SolidWorks 2016) 

Which version of the SolidWorks are you using?

gabe1294, 2017-08-04:
Hi Andreea,
I have newer versions of SOLIDWORKS installed.  2017 and 2018 Beta 2.  I
have tried loading files saved in earlier versions as well unsuccessfully.
Could you send me a solidworks file you know works?

Gabe

Ghostkeeper, 2017-08-20:
The two test files in here have often been tested: https://github.com/Ultimaker/CuraSolidWorksPlugin/tree/master/test_files

We've recently implemented support for SolidWorks 2017. It might help to uninstall the plug-in and re-install it (it might trigger a version update in the plug-in browser, I don't know).

pjo1976, 2017-10-26:
I am having the same problem with Cura 3.0.3.
I get the Invalid file (Even though it is a SW 1017 sldprt file) Failed to load and also the Error while starting Solidworks.
I have uninstalled, re-installed etc. but still the same error.

Also, the macro does not work within Solidworks, I get the Compile error, Cant find project or library

Any help on this would be much appreciated.

Paul

Ghostkeeper, 2017-11-05:
I'll just ping @thopiekar since he knows more about this and is able to test stuff since I don't have a SolidWorks installation :wink:

Perhaps you could share a simple sldprt file with us that gives the problem for you?

thopiekar, 2017-11-06:
You mean you reinstalled SolidWorks?
Well, then this is an known issue and the reason why the development of the plugin is standing still.

In order to repair macro editing on my laptop, I reinstalled SolidWorks and now the API is inconsistent. That means some functions are missing or returning wrong types of data.

Contacted the reseller of the licence for support, but after a while no progress on fixing this.
I guess the the case will soon be redirected to SolidWorks, so they will help from first hand. As soon as the fix is there and known I will let you know.

pjo1976, 2017-11-06:
I didn't uninstall/reinstall Solidworks, I uninstalled/reinstalled Cura, the plugin was working with Cura 2.7, then when I installed Cura 3.0.3 the plugin stopped working.

Thanks

Paul

LipuFei, 2017-11-06:
Hi @pjo1976  @gabe1294 , we have updated the SolidWorks plugin on the server, could you try updating your plugin through the plugin browser? If it doesn't work, could you also try uninstalling and installing the plugin again? Thanks!

pjo1976, 2017-11-06:
Hi,

I still get the same error
[image]
Thanks

Paul

LipuFei, 2017-11-06:
@pjo1976 You also need to restart Cura after the plugin update.

pjo1976, 2017-11-06:
I did!!

LipuFei, 2017-11-06:
Could you provide your log file? It's here: `C:\Users\<your-user-name>\AppData\Roaming\cura\3.0\cura.log`. Thanks

pjo1976, 2017-11-06:
Here you go!!

[cura.log](https://github.com/Ultimaker/Cura/files/1445768/cura.log)

LipuFei, 2017-11-06:
@pjo1976 I'm not sure what happened... but could you try one more thing for me?

1. Close Cura
2. Close your SolidWorks (if it's open)
3. Open Task Manager
4. In the "Processes" tab, look for the SolidWorks process(es). If there is one or more, end that task by right click -> "End Task".
5. Start Cura and try opening a SolidWorks file again.

[image]

thopiekar, 2017-11-08:
@pjo1970 , could you test 0.3.7 please?
Thanks!

- https://github.com/thopiekar/CuraSolidWorksPlugin/releases

pjo1976, 2017-11-09:
Hi Thopiekar,

All seems to be working again, can open files directly in Cura, and the macro in Solidworks is also working, exports and then opens in Cura.

Much appreciated.

Paul

thopiekar, 2017-11-09:
I love it 💪

thopiekar, 2017-11-09:
@pjo1976 If you have any feedback about the plugin, eg. missing features, then feel free to write me a mail. I can't guarantee that I'll be able to answer, but I will make at least notes about it and keep an eye on it.

thopiekar, 2017-11-09:
@gabe1294 How about you?

pjo1976, 2017-11-09:
Brilliant, much appreciated for the quick response and fix!!

gabe1294, 2017-11-09:
It's working for me now.  Not sure what the sequence was.  I'm running Cura
3.0.3.  Its working with SOLIDWORKS 2018 SP0.1
The Macro works also.

Thanks

thopiekar, 2017-11-09:
Great!
Thank you for your feedback. Also feel free if you see room for improvements.

Since it is working now for both of you, I'm closing this issue here.

closed 2017-11-09 by thopiekar