Ultimaker/Cura #2204: Crash on start

arka2pol, 2017-08-03:
Crash report below: (the only may be non standard operation on first run was not allowing network access in firewal)

Version: 2.7.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 707, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderStack.py", line 94, in _onPropertiesChanged
AttributeError: 'NoneType' object has no attribute 'definition'

-------------------------
Thread 0x00004568 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004564 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004544 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\serial\serialwin32.py", line 59 in open
  File "D:\env\2.7\inst\lib\site-packages\serial\serialutil.py", line 182 in __init__
  File "D:\env\2.7\inst\lib\site-packages\serial\serialwin32.py", line 31 in __init__
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\avr_isp\stk500v2.py", line 29 in connect
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDevice.py", line 315 in _connect
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004540 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDevice.py", line 339 in _connect
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004538 (most recent call first):
  File "C:\Program Files\Cura 2.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004534 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004530 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000452c (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
[... 60 lines cut ...]
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000044f0 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x000044d0 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 342 in showSplashMessage
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 652 in run
  File "D:/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

arka2pol, 2017-08-08:
Hm..... Today after re-downloading and re-installing it starts..... Other functions still not tested.

Ghostkeeper, 2017-08-20:
Fixed here: https://github.com/Ultimaker/Cura/commit/cf1309c650048ed027aabf32a0c0f9c028bfaedc

closed 2017-08-20 by Ghostkeeper