Ultimaker/Cura #2205: Missing support interface on shallow angles

jaibooot, 2017-08-03:
Hello, 

I am having a problem with missing interface layers when printing small interface layers. The interface builds fine on 90deg overhangs, but will not build on very shallow angles. Here is an example of a part with 5 overhangs (10, 20, 30, 40, and 50 deg). The overhang angle is set to 15 degrees. The settings are in the screenshot. The slicer builds interface on the 40 and 50 deg overhangs, but not in the 20 and 30 degree overhang. 

[image]

I believe the problem lies in the way intereface layers are calculated. It looks like the interface layers build in a constant z thickness. It seems like the desired behavior would be to have the interface build at a constant thickness at an offset normal to the overhang. 

[image]

Since my application necessitates thin interface thicknesses (.5mm or less), this commonly results in missing interface layers. Let me know if a change is possible. I have attached a sample part. 

Thanks for your help!
[Part Studio 1 - angle test.zip](https://github.com/Ultimaker/Cura/files/1198787/Part.Studio.1.-.angle.test.zip)

Jay

Ghostkeeper, 2017-08-21:
The support interface indeed has a constant Z thickness rather than a constant Euclidean thickness perpendicular to the normal. The normal is quite involved when surrounded by multiple surfaces, so Cura never uses it. The overhang angle is reproduced by multiplying the layer thickness by the tangent of the overhang angle so that a sort of "maximum overhang distance" is found. The lower layer is then offset by that distance and subtracted from the layer above it. Anything that remains will get support.
In short, we do offset magic to support only the bits of a layer that have more overhang than the specified angle.

Using a tilted cube I found the results that I think you're referring to. What happens is that the area that would get support interface is so thin (in the X/Y direction) that it's not possible to draw that area with the current line width.

Making it depend on the actual 3D distance is quite involved as this requires a 3D offset. 3D offsets are very hard to get right! It essentially requires us to pull in CGAL.

jaiboot, 2017-08-27:
Hey Ghostkeeper, 

Thanks for the detailed response! Is there any way other than a full 3d offset to make sure interface build on shallow angles with a small interface layer thickness?

ianpaschal, 2018-02-28:
@Ghostkeeper Can we estimate the likelihood of ever trying to make that improvement and assign a `Won't fix` or `Unplanned` accordingly?

jaibooot, 2018-04-21:
Could a minimum support interface size be enforced? This way, slight overhang angles would still build interface if support structures were present

Ghostkeeper, 2018-05-09:
I don't really see a good way of implementing that, and I also don't think it would work well.

We currently determine the places that need Support Interface by looking a certain number of layers above the support areas (interface thickness + z distance). For instance, let's say that interface thickness + z distance results in 7 layers. Then we take the support areas of layer X and take the portion of those areas where there is model on layer X+7. Those areas become the support interface.

To implement a "minimum support interface size" like setting, you'd have to gradually increase the interface thickness until this minimum support interface size is achieved? Or, maybe increase the interface thickness until there is any amount of support interface and then offset that area until the minimum area is reached?

This would crash if the user sets the area larger than the total support area (like, the user wants >3mm^2 interface but he loads a cube that doesn't need any support at all). And it won't work if you have a more complex model that has proper support interface on one size of the build plate but the problematic sort of steep angles on the other side.