Ultimaker/Cura #2206: XY corodinates of Move don't match home position of printer

PJParish, 2017-08-04:
The XY move coordinates don't start at 0 for X and Y where the home marker is located. The 0,0,0, point is the middle of the bed by default.  If a printer is selected with a home marker in the lower left corner (pretty much the standard for most printers) the coordinates of the position of the object don't match or get updated.

Ghostkeeper, 2017-08-21:
Yes, Cura has its own coordinate system with an origin in the centre of the build plate. This system is used for the transformations, such as the move tool.

There has been some confusion about Cura's coordinate system and the other various coordinate systems and all of their origins, which direction is Z and such with various printers holding different standards, circular build plates, CAD systems doing their own thing, etc. I think we should only display printer-coordinate positions so that the result can be verified in the g-code.

Ghostkeeper, 2018-01-01:
Our project manager removed this from our planning because it's older than 12 weeks.

GregValiant, 2025-07-03:
There have been recent changes to the "transformation" and "rotate" tools.  There is now an option in the "Preferences" to make the direction of the model coordinate "Y" match the Cura build plate "Y".  That's as far as it went.

It's been 8 years so I'll go ahead and close this.  If you have strong feelings then it can be re-opened.

closed 2025-07-03 by GregValiant