Ultimaker/Cura #2209: Ooze shield with small brim or skirt attached

PJParish, 2017-08-04:
Since an ooze shield is only a single line wide there are times when it will not stay attached to the bed very well.  It is catching a lot of ooze strings and can easily get ripped off the bed.

I turn on the skirt with a distance of 2 and 4 lines to create a "Brim" that the ooze shield connects to to keep it on the bed better.  This only works on the outside shields.  Inside shields don't get a skirt attached.

If I turn on a full brim as a workaround it gets attached to the object as well which I'm trying to avoid.

If an ooze shield could automatically have the bottom layer be 4 lines thick (or user selectable?) to the outside of it (side away from the object) this would make it much easier to set it up and have it stick well.

Ghostkeeper, 2018-01-01:
It still works with the brim.

Improving the ooze shield adhesion on inside by attaching it to the skirt was just removed from our planning by our project manager because it was older than 12 weeks.

Schleichmichl, 2020-01-06:
I have skirt enabled but the ooze shield is not strengthend at all locations.
Unfortunately, it looses contact where the arrows indicating.

Can I do anything else? Is there a workaround?
[image]

Ghostkeeper, 2020-01-13:
You can print with a brim instead?

Schleichmichl, 2020-01-13:
One could as a workaround. But: Perhaps you don't want to use brim for aesthetic reasons.
Actually, this is often the only reason for using the ooze shield. 
My prints are quite good without ooze shield, but when I what an aesthetic object, I don't want to cut the brim off :-(

PJParish, 2020-01-14:
Schliechmichl,

Not wanting to put a brim around everything I was printing was the original reason I made the request for an ooze shield only brim.  As you have seen skirt only does not cut it.  I have not come up with a good way around this using Cura when a multi-color multi-part assembly has inside and outside contour ooze shields and I don't want a brim connecting all the print in place assembly parts together.

Ghostkeeper, is it worth trying to get this put back on the list of improvements?  Ooze shield with automatic 4 line brim.

Ghostkeeper, 2020-01-14:
I'll propose it again to the team. Not giving it a lot of hope though, because the ooze shield is not a particular area of focus for us now.

hbh7, 2020-03-31:
Just starting with dual extrusion and saw the ooze shield easily peel up after only a few layers of printing... Sad to hear that there's no way to turn on a brim for it as that was my first thought. Seems like it could be exactly the same as the prime tower has.