Ultimaker/Cura #2210: launch cura app successfully before but now crashed

yingzhang1109, 2017-08-04:
I installed all cura files and set the environment about two months ago. It succeeded and I could launch cura through my python interpreter. A few days ago, I uninstalled the python by mistake and reinstalled python as before. unfortunately, it's not working right now. Cura stops before open the main application. Does any one have any idea about that? Thanks in advance

yingzhang1109, 2017-08-04:
that's the log file 

2017-08-04 17:19:26,627 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin FileLogger
2017-08-04 17:19:26,635 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GCodeProfileReader
2017-08-04 17:19:26,640 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GCodeReader
2017-08-04 17:19:26,643 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GCodeWriter
2017-08-04 17:19:26,648 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ImageReader
2017-08-04 17:19:26,656 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LayerView
2017-08-04 17:19:26,661 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LegacyProfileReader
2017-08-04 17:19:26,666 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LocalFileOutputDevice
2017-08-04 17:19:26,670 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MachineSettingsAction
2017-08-04 17:19:26,677 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-08-04 17:19:26,677 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin PerObjectSettingsTool
2017-08-04 17:19:26,683 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin PostProcessingPlugin
2017-08-04 17:19:26,702 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin RemovableDriveOutputDevice
2017-08-04 17:19:26,717 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SliceInfoPlugin
2017-08-04 17:19:26,722 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SolidView
2017-08-04 17:19:26,727 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CameraTool
2017-08-04 17:19:26,733 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MirrorTool
2017-08-04 17:19:26,752 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin RotateTool
2017-08-04 17:19:26,760 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ScaleTool
2017-08-04 17:19:26,764 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SelectionTool
2017-08-04 17:19:26,771 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin TranslateTool
2017-08-04 17:19:26,779 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UltimakerMachineActions
2017-08-04 17:19:26,959 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UM3NetworkPrinting
2017-08-04 17:19:27,063 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UpdateChecker
2017-08-04 17:19:27,075 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2017-08-04 17:19:27,176 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin USBPrinting
2017-08-04 17:19:27,190 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin VersionUpgrade21to22
2017-08-04 17:19:27,199 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin VersionUpgrade22to24
2017-08-04 17:19:27,208 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SimpleView
2017-08-04 17:19:27,213 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin WireframeView
2017-08-04 17:19:27,252 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin X3DReader
2017-08-04 17:19:27,260 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin XmlMaterialProfile
2017-08-04 17:19:27,270 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin XRayView
2017-08-04 17:19:27,271 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFReader
2017-08-04 17:19:27,272 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-08-04 17:19:27,272 - ERROR - UM.Logger.logException [78]:   File "C:\Users\Desktop\cura-backup - 1 - backup\UM\PluginRegistry.py", line 293, in _findPlugin
2017-08-04 17:19:27,272 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-08-04 17:19:27,272 - ERROR - UM.Logger.logException [78]:   File "D:\python3.5\lib\imp.py", line 244, in load_module
2017-08-04 17:19:27,273 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-08-04 17:19:27,273 - ERROR - UM.Logger.logException [78]:   File "D:\python3.5\lib\imp.py", line 216, in load_package
2017-08-04 17:19:27,273 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-08-04 17:19:27,273 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-08-04 17:19:27,274 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-08-04 17:19:27,274 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module
2017-08-04 17:19:27,274 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2017-08-04 17:19:27,274 - ERROR - UM.Logger.logException [78]:   File "C:\Users\Desktop\cura-backup - 1 - backup\UM\..\plugins\3MFReader\__init__.py", line 5, in <module>
2017-08-04 17:19:27,274 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFReader
2017-08-04 17:19:27,275 - ERROR - UM.Logger.logException [78]:   File "C:\Users\Desktop\cura-backup - 1 - backup\UM\..\plugins\3MFReader\ThreeMFReader.py", line 22, in <module>
2017-08-04 17:19:27,275 - ERROR - UM.Logger.logException [78]:     import Savitar
2017-08-04 17:19:27,275 - ERROR - UM.Logger.logException [78]: ImportError: No module named 'Savitar'
[... cut ...]
2017-08-04 17:19:37,926 - DEBUG - UM.Controller.setActiveView [81]: Setting active view to SolidView
2017-08-04 17:19:39,741 - DEBUG - UM.Qt.Bindings.Theme.load [126]: Loading theme file: C:\Users\Desktop\cura-backup - 1 - backup\cura\..\resources\themes\cura\theme.json
2017-08-04 17:19:39,844 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme C:\Users\Desktop\cura-backup - 1 - backup\cura\..\resources\themes\cura
2017-08-04 17:19:40,219 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [182]: Could not get metadata of container  because it was not found.

Ghostkeeper, 2017-08-22:
What's your Python version now with the new installation?

Could you run `python --version`?

Ghostkeeper, 2017-08-22:
It's also giving a lot of reports that it can't find Savitar. Perhaps you have to `make install` again on libSavitar to install the Python bindings again?

ianpaschal, 2018-03-02:
Needs info but never got it. Closing.

closed 2018-03-02 by ianpaschal