Ultimaker/Cura #2215: Slice time for multiple parts is N*log(N) and not N - a bug somewhere?

hoegge, 2017-08-05:
When you make multiple copies of the same part on the plate the slicing time PER PART increases with apx. Log N. So the time it takes should be proportional to N and not NlogN. Actually, since Cura knows they are copies, Cura should only slice them once and just translate the g-code. So the time should be even faster for copies.

Result of slicing different numbers of a smal part (cylindrical disc with a hole):

```
Items | Total time | Time per item
1 | 1.18 | 1.18
2 | 1.81 | 0.91
4 | 3.31 | 0.83
8 | 5.71 | 0.71
16 | 15.98 | 1.00
24 | 36.06 | 1.50
32 | 60.75 | 1.90
40 | 96.72 | 2.42
48 | 145.03 | 3.02
56 | 204.91 | 3.66
```
[image]

Ghostkeeper, 2017-08-23:
Actually it increases quadratically in the worst case, in theory (in practice there are so many factors involved that it becomes different with every model).

This is because of our planning which finds the nearest object to print next with a simple selection sort (i.e. nearest neighbour).

Ghostkeeper, 2017-08-23:
We can't just copy-paste multiple parts of g-code because of two reasons:

* We don't know that they are the same. As far as the slicer sees, the models aren't exactly the same, since there are rounding errors due to the transformation. Even if these rounding errors were not present, trying to match pieces that are exactly the same would be prohibitively expensive to compute.
* The resulting g-code is not the same since the print head is coming from a different direction initially, which makes the order of the lines completely different.

ianpaschal, 2018-02-28:
@Ghostkeeper Can we estimate if this is a change we plan to address and assign `Won't Fix` or `Unplanned` accordingly? Since print time estimation is preeeeetty darn good almost all of the time I'm leaning towards calling this a non-issue.

hoegge, 2018-02-28:
It has nothing to do with Print time - it is Slicing time - and that it takes more and more time PER ITEM to slice several identical items

ChrisTerBeke, 2018-03-02:
@Ghostkeeper explains exactly that in his answer above.

closed 2018-03-02 by ChrisTerBeke