Ultimaker/Cura #2216: Cura 2.6.1 fails to slice stl correctly

RLuckom, 2017-08-05:
This is a screenshot of the way Cura displays the STL I am slicing. Note that there is a rectangular cutout in the face of the cube towards the view frame, but that the cutout does *not* extend through the opposite face of the model.

[image]

When I click 'Save to file' and generate a gcode file, the resulting file is different than the displayed model. Specifically, the cutout in the picture above extends through the entire model. Using the online gcode viewer at gcode.ws to take the same screenshot shows the following, which is how the model actually gets printed:

[image]

I'm happy to provide all my source and model files to help reproduce this issue, but don't see a way to attach them here.

Environment: 
OS: Ubuntu 16.04
STL generated by OpenScad version 2015.03-1 
Cura 2.6.1

smartavionics, 2017-08-06:
How thick is the wall that is missing? If it is thinner than the line width then it's not going to be printed. You can attach files to this issue by following the instructions at the bottom of the text entry box that I am typing into right now! I use openscad so please just attach that and I will take a look at your model.

RLuckom, 2017-08-06:
I did try to use that method for attaching files (it's how I attached the images ;-) ) But that dialog rejects non-image files unfortunately. Here is my openscad code as requested.

     SCREW_SHAFT_DIA = 4.1;
     SCREW_HEAD_HEIGHT = 2.54;
     SCREW_HEAD_DIA = 7.7;
     NUT_FLAT_TO_FLAT = 8.52;
     NUT_POINT_TO_POINT = 9.68;
     NUT_THICKNESS = 2.97;
     FN = 60; 
     SCREW_OUTLINE_SCALE_FACTOR = 1.03;
    
     module screw(shaft_length_in) {
         shaft_length = (25.4 * shaft_length_in) + 1;
         union() {
         cylinder(SCREW_HEAD_HEIGHT, d=SCREW_HEAD_DIA, $fn=FN);
         translate([0, 0, SCREW_HEAD_HEIGHT - 1]) cylinder(shaft_length,
     d=SCREW_SHAFT_DIA, $fn=FN);
         }   
     }   
    
     module hex_nut() {
         cylinder(NUT_THICKNESS, d=NUT_POINT_TO_POINT, $fn=6);
     }   
    
     module hex_nut_pocket(depth) {
         cube([depth, NUT_FLAT_TO_FLAT, NUT_THICKNESS]);
     }   
    
     module screw_recess(shaft_length_in) {
         scale([SCREW_OUTLINE_SCALE_FACTOR, SCREW_OUTLINE_SCALE_FACTOR, 1]) 
     screw(shaft_length_in);
     }   
    
     module test_piece() {
         height = 15; 
         width = 12; 
         center = width / -2; 
         difference() {
             translate([center, center, 0.001]) cube([width, width, height]);
             union() {
                 screw_recess(.5);
                 translate([0, 0, (height - (NUT_THICKNESS -0.01))]) hex_nut();
                 translate([center + .001, NUT_FLAT_TO_FLAT / -2, SCREW_HEAD_HEIGHT +
     4]) hex_nut_pocket(12);
             }   
         }   
     }   
    
     test_piece();
     //translate([-6, NUT_FLAT_TO_FLAT / -2, SCREW_HEAD_HEIGHT + 2]) 
     //hex_nut_pocket(12);

module hex_nut_pocket(depth) {
    cube([depth, NUT_FLAT_TO_FLAT, NUT_THICKNESS]);
}

module screw_recess(shaft_length_in) {
    scale([SCREW_OUTLINE_SCALE_FACTOR, SCREW_OUTLINE_SCALE_FACTOR, 1]) screw(shaft_length_in);
}

module test_piece() {
    height = 15;
    width = 12;
    center = width / -2;
    difference() {
        translate([center, center, 0.001]) cube([width, width, height]);
        union() {
            screw_recess(.5);
            translate([0, 0, (height - (NUT_THICKNESS -0.01))]) hex_nut();
            translate([center + .001, NUT_FLAT_TO_FLAT / -2, SCREW_HEAD_HEIGHT + 4]) hex_nut_pocket(12);
        }
    }
}

test_piece();

smartavionics, 2017-08-06:
Hi, yeah, you have to attach that as a text file (.txt). When I view it with openscad I can see that the cutout extends all the way to the far wall so I would not expect cura to have a wall there.

[image]

If you made that cutout very very slightly wider so that the volume you are subtracting extends through the outside of the volume you are subtracting from, openscad would show you the hole rather than just that shaded region.

RLuckom, 2017-08-06:
Thank you! I will close this.

closed 2017-08-06 by RLuckom