Ultimaker/Cura #2218: Not working correctly on scaled display since v. 2.6.2

hyteoo, 2017-08-06:
OS: Windows 10
Display scaling: 200%

Since 2.6.2 it seems like the interface is "shifted" slightly higher. Moving the mouse over the interface is very tricky, mouse pointer has to be lower than elements to actually interact with them.

The menu is also not visible anymore, the interface seems to be shifted by the height of the menu.

fieldOfView, 2017-08-07:
This has been reported once or twice before and was solved by updating the graphics driver

hyteoo, 2017-08-16:
It doesn't seem to be drivers, I keep up to date with drivers (Dell XPS 15 9550)
Reverting back to older versions (2.6 for ex) works fine
It's something in the 2.6.2 version
Alos checked with 2.7beta, same problem

Ghostkeeper, 2017-08-23:
Duplicate of #2068.

closed 2017-08-23 by Ghostkeeper