Ultimaker/Cura #2219: Cura won't slice anything. Tried every possible solution to my knowledge

RedDropOriginal, 2017-08-06:
Recently I fired up Cura 2.6.1, never had any issues before today. but now it will 'slice' the model but the estimates of time and weight are just zero. The gcode generated is just start and end-script. 
It happened after I wanted to try the spiralise mode, it didn't really work out so I unchecked it and now I have this going on.
I tried:
→ Different model
→ Different profile/material
→ Restarting Cura/PC
→ Checking and unchecking every setting

[image]

RedDropOriginal, 2017-08-06:
Here are all the settings I use: 
[image]
[image]
[image]
[image]

Vandrasc, 2017-08-07:
I can't seem to reproduce this one, you tried removing your user directory folder?

Ghostkeeper, 2017-08-23:
Doesn't reproduce for me either. I copied over all the settings and it just slices my Benchy.

nickthetait, 2017-08-23:
@RedDropOriginal, could you try the same test but using the latest beta (2.7.0) release?

nallath, 2017-08-23:
Alternatively; Share a project file with us.

RedDropOriginal, 2017-08-23:
@nallath what type of file is this? How do I shzre a project file?

RedDropOriginal, 2017-08-23:
@Vandrasc @Ghostkeeper @nickthetait I reinstalled cura. Tried new versions, nothing helps. But I did find out that when I switch to a different printer in cura everything works fine... I matched the printer settings of both machines, same start/end scripts and everything, but still my original machine that I had first won't work...

nallath, 2017-08-23:
file->save project

ianpaschal, 2018-02-28:
`Needs info` but never got it + irreproducible. Closing.

closed 2018-02-28 by ianpaschal