Ultimaker/Cura #2221: 64bit or 32 bit?

yingzhang1109, 2017-08-07:
I found that, in the ultimaker website, there is no 32 bit version after cura 2.4. is it the same here? no 32 bit version anymore?

Ghostkeeper, 2017-08-07:
No, we don't create a 32-bit version any more, since we found that there are very few users with a 32-bit computer and it was taking considerable maintenance to produce that build.

closed 2017-08-23 by Ghostkeeper

spitstec, 2018-03-25:
I run an old version of cura (15.04.6) on 32 bit windows computers. Lately I get a "new update available" messages on startup. When I choose to install a new update it downloads a 64 bit version which i cannot run. Is there a way to get rid of this annoying  "new update available" without installing the for me useless 64 bit version?

fieldOfView, 2018-03-25:
See File -> Preferences -> Cura settings -> Check for updates