Ultimaker/Cura #2225: A fatal exception has occurred

JanBeckman, 2017-08-08:
Version: 2.7.0-BETA
Platform: Darwin-16.7.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 707, in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 261, in event
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315, in __performEmit
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderStack.py", line 94, in _onPropertiesChanged
AttributeError: 'NoneType' object has no attribute 'definition'

-------------------------
Thread 0x000070000a51f000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000a01c000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009a96000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009593000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009090000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008b8d000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000868a000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008187000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
[... 45 lines cut ...]

Thread 0x00007000058ec000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fffd81543c0 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 676 in run
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

jonathanweinberg, 2017-08-09:
I'm getting a similar crash:

```
Version: 2.7.0-BETA
Platform: Darwin-16.6.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 73, in <module>
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 646, in run
AttributeError: 'NoneType' object has no attribute 'setOrigin'

-------------------------
Thread 0x0000700012a7a000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700012577000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700011ff1000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700011aee000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000115eb000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000110e8000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap
[... 76 lines cut ...]
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000d4c4000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
[... cut ...]
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fffadc563c0 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
```

Vandrasc, 2017-08-09:
Yes, this will be fixed in 2.7, thank you for reporting it.

Vandrasc, 2017-08-09:
@jonathanweinberg could you please share your logs with us? Your error is caused by something different. Thanks

Ghostkeeper, 2017-08-23:
Duplicate of #2204.

closed 2017-08-23 by Ghostkeeper