Ultimaker/Cura #2226: Cura 2.6.2 Not Responding

CookedChili, 2017-08-09:
Hello,

I have been using Cura 2.6.2 on Windows 10, but whenever I try to use it on my other monitor, the splash screen shows up, but the program itself no longer responds. I have tried reinstalling and every version since Cura 15 has the same result. Is there any way to fix this?

Here is the splash screen before loading in:
[image]

Cura 2.6.2 not responding:
[image]

The latest line in my log file reads:
2017-08-01 05:40:06,987 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [T:\Program Files\Cura 2.6\resources\shaders\platform.shader]...

I have made sure that all of my drivers (for intel or nvidia) are updated, which they were already.
Running Windows 10 on Intel i7-6850 with a GTX 1080 GPU. Latest Intel driver as of 8-8-2017.

For more information, I have posted on the Ultimaker Forums, here: https://ultimaker.com/en/community/50456-cura-262-not-responding

CookedChili, 2017-08-15:
If someone has an answer or a better solution, please let me know!!

Ghostkeeper, 2017-08-23:
We aren't able to reproduce this on any of our testing computers, so it's hard to find out what is going wrong :(

Ghostkeeper, 2017-08-29:
Also doesn't happen on my home computer (with 2 screens of different resolutions).

CookedChili, 2017-09-09:
I have 2 monitors, both at the same resolution (3860 x 2140), and the issue keeps happening with Cura 2.7. No response from the program, the window is white, and then it automatically closes, as if it crashed. This only happens when I run it on my alternate monitor, but when I want to open it again, I can't and have to completely uninstall Cura along with my profiles that I created.

CookedChili, 2017-09-09:
I hope that helps, I don't know if I am doing anything wrong :(

fieldOfView, 2017-09-09:
@CookedChili, totally uninstalling and removing Cura is not necessary. Here's a subtler way to resuscitate Cura:
* Open the start menu and type in %APPDATA%\cura\2.7
* Press enter
* Remove cura.cfg (only)
* Restart Cura
Cura will now ask you to add a new printer. Cancel out of that dialog, and reselect your previous printer from the dropdown on top of the sidebar.

Ghostkeeper, 2017-09-15:
That would at least reset the window position to the default (which would be on the main screen) but you have to be very careful not to move Cura to the other screen then!

Maybe CookedChili has two different graphical cards, one for each of his monitors, and then Cura's shaders don't render well on the one for the alternate monitor?

CookedChili, 2017-10-22:
I only have a GTX 1080, and I have both monitors running on it. I don't know if there is any other issue?

Ghostkeeper, 2017-10-31:
I wouldn't know then... Maybe one of the monitors has a different pixel scale than the other? Is one of them High-DPI?

CookedChili, 2018-01-11:
Sorry for the late reply; I have 2 of the exact same monitors, configured the exact same way. I have made sure that there is no High-DPI. Still not fixed with Cura 3.1.0

jackha, 2018-01-15:
@CookedChili Possibly a stupid question, but did you try with a single monitor? Does it work fine then?

Ghostkeeper, 2018-01-28:
Cura normally supports dual monitors. Here's a screenshot showing that off on my desktop, where I have different resolutions:
[image]

It might not be related to having two monitors but something else that crashes just after that shader is loaded. That's when the GUI is shown for the first time, after all.

fieldOfView, 2018-01-28:
> It might not be related to having two monitors

Dual GPU is a much likelier culprit.

Ghostkeeper, 2018-01-28:
Yeah, but CookedChili mentioned:

> I only have a GTX 1080, and I have both monitors running on it.

btunnadine, 2018-07-25:
I believe I have this problem as well. Possibly the same as #3165 as well? 

**Application Version**
I am using version 3.4.1, but it also occurred on 3.3.0

**Platform**
Windows 10

**Printer**
Monoprice Select Mini v2 (E3D)

**Steps to Reproduce**
A) Run Cura when either the config states to display on secondary monitor
   -or-
B) When Cura is running, move the application to Monitor 2 and attempt to use any dropdown UI element (top menu, solid/X-ray/Layer view, material setup etc.).

**Actual Results**
A) Program does not run
   -or-
B) Program immediately crashes to desktop

**Expected results**
A) Program runs
   -or-
B) Dropdown menu opens

**Additional Information**
I am using a dual-gpu setup as suggested by fieldOfView, main monitor is an Acer Predator XB271HU at 1440p 144hz running on AMD Radeon R9 390 GPU, secondary is an LG IPS224 1080p 60hz running on onboard Intel HD Graphics 4000.
I have used a set-up with the LG monitor as my main monitor (GPU) and the secondary as a 1680x1050 60hz Hanns-G monitor (onboard), which still exhibited the same behaviour.

This also occurs if only the secondary monitor is connected (on Intel HD Graphics), and does not if only the primary is connected (on Radeon R9 390).

In the previous version (3.3.0) the splash screen would always appear on the main monitor when starting the app on the second monitor, but after it had finished loading it would crash. If the app is started from the main monitor and the config states to open on the second (window_left + window_width causes it to be more on the second than on the first I think?) it crashes like this as well.

I can move/rotate the model around inside the 3d viewport; switch to and use the monitor (with octoprint plugin); and use any other UI button, checkbox, tooltip etc., but dropdown menus, new windows (created by printer configuration etc.) will crash the application if it is on the second monitor or if the new window is created on the second monitor.

Overall this only poses a minor inconvenience in most cases, but can prove annoying if plugin or config windows attempt to open on the second display.

Log file: [cura.log](https://github.com/Ultimaker/Cura/files/2229701/cura.log)

dmorris68, 2018-11-23:
Just came to post that 3.6 has this issue the same thing with my triple monitor setup. I have a 4K center monitor flanked by 1440p monitors. The 4K and right 1440p monitor are run from a Nvidia GTX 1070. The leftmost 1440p monitor runs off the onboard Intel UHD 630 GPU. My Windows desktop is on the center 4K (Nvidia) monitor, where I launch Cura. If I then move Cura to **either** of the other monitors, it becomes unstable and eventually hangs. It would kinda make sense that the shader gets confused crossing GPUs (although it should detect this and reset itself), but it also doesn't seem to like moving to a different resolution.

CookedChili, 2018-11-23:
I don’t mean to be critizing anyone who is developing Cura, but the issue has been present since Cura 15. Someone at Ultimaker has to have 2 monitors. To replicate it, all you need to do, at least for me is the following:
1. Open Cura (any version, haven’t tested 3.6 yet but based on prior comments still doesn’t work)
2. Drag Cura so that it ONLY is on the secondary monitor. 
3. Try printing something or at least slicing and exporting something.
4. While having it ONLY on the other monitor close it.
5. When you reopen Cura after having it closed for quite some time, you will have the splash screen appear, but the main window is blank with absolutely nothing there.

ianpaschal, 2018-11-23:
> Someone at Ultimaker has to have 2 monitors.

Pretty much all of us have 2-3 monitors but this issue does not occur for anyone in the building.

> Please don’t close issues that have not been resolved.

Thanks, but we'll manage our tickets as we see fit. There absolutely does not need to be a 1:1 relationship between "closed" and "resolved." We "reject" tickets all the time because we're unable to reproduce them, don't have time to work on them, or don't consider them a good use of the time we do have. Also closed ≠ deleted; we can reopen an issue if we change our mind.

That being said, I'm really sorry you're experiencing this issue, but no one within our company is experiencing it, despite almost everyone working with external monitors. Probably it's because we don't use exactly the same graphics cards and monitors as you do, but the unfortunate reality is that this means unless you ship us your computer, we really can't troubleshoot it.

nallath, 2018-12-04:
We have spent quite a bit of time on this, it's really not like we're dismissing it because we don't feel like putting the time in.
There is a quite wide spread of computers / GPU's & operating systems being used. We even keep a number of different configurations specifically for this kind of testing. But unfortunately, none of these seem to exhibit the problem.

achugb, 2020-07-29:
https://github.com/Ultimaker/Cura/issues/7848#issuecomment-665721836