Ultimaker/Cura #2227: Scannect created files fail to load on the 2.6.2, loads fine on 2.1.3

robotikosmokykla-lt, 2017-08-09:
[Ricardas.zip](https://github.com/Ultimaker/Cura/files/1210516/Ricardas.zip)

smartavionics, 2017-08-09:
Meshlab wouldn't open it saying "premature end of file". Fortunately, FreeCAD not only opened it but fixed up a large number of problems with the mesh. That mesh is a mess! I attached the fixed version.

[Ricardas-fixed.zip](https://github.com/Ultimaker/Cura/files/1210691/Ricardas-fixed.zip)

[image]

robotikosmokykla-lt, 2017-08-09:
Thanks, will try fixing with netfabb. Though strange that the 2.1.3 opens
it.

fieldOfView, 2017-08-09:
I think 2.1.3 parsed STL files in Python, whereas later version use Numpy-STL, a third party module. This module is much, much faster (around 10x), but apparently somewhat less tolerant of faulty meshes.

Ghostkeeper, 2017-08-23:
Today we updated to a newer version of Numpy-STL. Perhaps that fixes things, because we updated it due to some of the build platform models not loading.

It was not updated in the 2.7 version though, only for releases after that.

closed 2018-03-02 by ChrisTerBeke