Ultimaker/Cura #2234: change material, determine percentage

maaspt, 2017-08-09:
Is there any function in Cura that can stop printer in determine percentage to change filament color ?

If we have only one extruder in printer .

I can do this in printer menu but I must control myself percentage to stop printer and change it.

I just need to chander layers colors in some projects.

Regards

maaspt

fieldOfView, 2017-08-10:
Check out Extensions -> Post Processing -> Modify G-code. Here you can add a "Pause at Height" script.

maaspt, 2017-08-10:
Hi
It works great for me like that.

Thanks for your help.

closed 2017-08-23 by Ghostkeeper

maaspt, 2017-09-05:
Hi

I upgraded to two extruders, one nozzle, how to change extruder in determine layer/percentage ?

In Post processing I can Pause at Heigth, and change filament manually in the same extruder, but with two extruder is there any function to change them auto ?

Regards