Ultimaker/Cura #2236: infill not showing in preview

fernx, 2017-08-10:
infill not showing in preview
[image]

fieldOfView, 2017-08-10:
What is your infill percentage? Is it by any chance 100? An infill percentage of 100 actually results in no infill, but all layers being considered as top layers.

fernx, 2017-08-10:
The infill is at 25%

RedDropOriginal, 2017-08-11:
do your models kinda look like this by any chance?

[image]

fernx, 2017-08-12:
Print without the infill and have 25% of cubic infill.

fernx, 2017-08-12:
[image]

Ghostkeeper, 2017-08-23:
So it's not just the layer view then but also the actual print that's missing infill?

ianpaschal, 2018-02-27:
Since this needs info but the OP did not respond for 6+ months, I'm closing this issue.

closed 2018-02-27 by ianpaschal

Ghostkeeper, 2018-03-21:
We'd need some information about what settings he was using if the actual infill is missing.