Ultimaker/Cura #2238: Cura 2.6.2 and 2.7 beta hang on startup under OS X

kaimikael, 2017-08-10:
I just installed Cura 2.6.2 on my MacBook, running OS X 10.10.5.  At startup I get the splash screen, and then it just hangs there and I have to kill the process.  In the log I get messages like:
`2017-08-10 11:56:08,000 kernel[0]: process cura[75501] caught causing excessive wakeups. Observed wakeups rate (per sec): 1721; Maximum permitted wakeups rate (per sec): 150; Observation period: 300 seconds; Task lifetime number of wakeups: 45374`

I then tried to upgrade to 2.7 beta, with the same hanging, but I also got the attached diagnostic reports which yet again warn about excessive wakeups.
[cura.zip](https://github.com/Ultimaker/Cura/files/1214549/cura.zip)

Ghostkeeper, 2017-08-23:
Could you share with us also the Cura log? You should be able to find it here: `$User/Library/Application Support/cura/2.7/cura.log`

kaimikael, 2017-08-24:
Attaching the log file here.  Also, here is another clue: The problem seems to be dependent on what devices are attached to my computer—after I unplugged my USB-connected telephone, Cura started nicely.  If the phone is connected, Cura hangs at the message “Loading machines…” on the splash screen.

[cura.zip](https://github.com/Ultimaker/Cura/files/1248035/cura.zip)

Ghostkeeper, 2017-08-31:
Here is the relevant snippet of the log:

    2017-08-24 09:11:13,429 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [311]: Attempting to connect to /dev/cu.usbmodem1452
    2017-08-24 09:11:13,430 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [311]: Attempting to connect to /dev/cu.usbmodem1454
    2017-08-24 09:11:13,454 - DEBUG - UM.Backend.Backend._logSocketState [173]: Socket state changed to Listening
    2017-08-24 09:11:13,541 - INFO - UM.Backend.Backend.startEngine [74]: Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
    2017-08-24 09:11:13,542 - DEBUG - UM.Backend.Backend._backendLog [92]: [Backend] Calling engine with: ['/Applications/Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/Applications/Cura.app/Contents/MacOS/resources/definitions/fdmprinter.def.json', '']
    2017-08-24 09:11:13,544 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/cu.usbmodem1452 to remove
    2017-08-24 09:11:13,546 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/cu.usbmodem1454 to remove

Ghostkeeper, 2017-08-31:
I've made a small fix that hopefully prevents this: 3951239513dd4ae341583e4e629b5b0876474c18

Ghostkeeper, 2017-08-31:
Let's close this once kaimikael confirms that it's fixed.

The fix will be included in the next beta in 3 weeks.

kaimikael, 2017-08-31:
OK, will download in three weeks and let you know then.

kaimikael, 2017-09-04:
Ah, there seems to be an official 2.7.0 release on the web site.  I downloaded it and while the startup is slow, it eventually passes the “loading machines” stage and starts up properly.  
Thanks for the fix!

closed 2017-09-04 by kaimikael

Ghostkeeper, 2017-09-10:
Uh, my fix wasn't included in the 2.7 stable release! Only in master, so it would be for the 3.0 beta...

Glad it worked again for you though.

commit 39512395 (2017-08-31): Don't emit devices changed signal if remove wasn't successful

The same could be done for adding but let's keep it at this for now.

Hopefully fixes #2238.